mne-tools/mne-python #9641: Regression in "How to plot topomaps the way EEGLAB does" example - montage.get_positions() returns different positions than the channel positons in evoked object.

mmagnuski, 2021-08-04:
Hi, I just noticed that the [How to plot topomaps the way EEGLAB does](https://mne.tools/dev/auto_examples/visualization/eeglab_head_sphere.html) example no longer shows correct EEGLAB-like channel layout:
[image]
the two layouts are the same on the current development and stable versions (0.23), which is incorrect. On version 0.22 these layouts are different (and correct):
[image]

The sphere parameters used in the example are not the same as before and this seems due to a change in how channel positions are obtained. Previously they were dug out from `.info['chs']` dictionary, now `montage.get_positions()` is used. However these two are not exchangable because the applied montage is in head coordinates, while the positions returned by `.get_positions()` show 'unknown' coord_frame.
In the same example doing:
```python
pos_mntg = biosemi_montage.get_positions()['ch_pos']['Oz']
ch_idx = fake_evoked.ch_names.index('Oz')
pos_evkd = fake_evoked.info['chs'][ch_idx]['loc'][:3]

print([f'{v:0.5f}' for v in pos_mntg])
print([f'{v:0.5f}' for v in pos_evkd])
```
gives different positions:
```
['0.00000', '-0.09494', '-0.00332']
['0.00000', '-0.09494', '0.03683']
```
I'm not sure what the best solution should be - is the intent of `montage.get_positions()` is to give channel positions in unknown coord frame? If so, the example should be reverted to use the old and ugly `.info['chs'][idx]['loc'][:3]` way of obtaining channel positions.

drammock, 2021-08-04:
cc @alexrockhill any chance you know what caused this?

alexrockhill, 2021-08-04:
Hmmm, I'm not sure, @mmagnuski is right that `montage.get_positions()` is different but it should just add a roundtrip operation. To my knowledge what happens in the example is that the montage is transformed to "head" coordinates using `mne.channels.montage._get_montage_in_head` and then stored in `info['chs'][...]['loc']`. I would guess what's going wrong is that the default biosemi coordinate frame is "unknown" so this transform is doing `mne.transforms.get_ras_to_neuromag_trans` to change to "head" when it is actually already in "head". I'll try a quick PR and see if that fixes it.

alexrockhill, 2021-08-04:
Hmmm changing the default to head makes them both look like the spaced out EEGLAB version 0.22 montage.

alexrockhill, 2021-08-04:
Actually one thing to try might be `fake_evoked.get_montage().get_positions()` because that uses a montage that has been transformed to "head".

mmagnuski, 2021-08-04:
@alexrockhill That makes sense (your previous post about changing default to head) as MNE (as far as I know) defines head coordinates with respect to fiducial positions (z = 0), while the biosemi positions are likely in a coordinate system where Oz is at z=0 (which is exactly what the example tries to do - move the sphere origin a bit higher to be at the z level of Oz channel).

alexrockhill, 2021-08-04:
Yep, that fixed it. Making a PR.

PR #9644 (2021-08-04): [BUG, MRG] Fix biosemi example coordinate frames
Fixes #9641.
closed 2021-08-04 by mmagnuski