mne-tools/mne-python #9649: MAINT, CI: pyvista and scipy deprecation warnings

drammock, 2021-08-05:
https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=14743&view=logs&j=fa98c7b3-575e-55b6-6946-22dea6f6daf8&t=c0866e65-eed5-5461-eadc-f0ff150eefd8&l=3422

> DeprecationWarning: rcParams is deprecated.  Please use ``pyvista.global_theme``

The CI installed pyvista 0.31.3 (latest release, from 23 June)

Note that there are also SciPy deprecation warnings appearing for `pinvh`:

https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=14743&view=logs&j=fa98c7b3-575e-55b6-6946-22dea6f6daf8&t=c0866e65-eed5-5461-eadc-f0ff150eefd8&l=3154

> DeprecationWarning: Use of the "cond" and "rcond" keywords are deprecated and will be removed in future versions of SciPy. Use "atol" and "rtol" keywords instead

cc @GuillaumeFavelier  do you know the right fix for the pyvista one off the top of your head?  If not I can try to sort out both of these tomorrow (unless someone else beats me to it).

GuillaumeFavelier, 2021-08-06:
Is it an old version of mne?

This is fixed in `main` by the new version of `_disabled_depth_peeling()`:

https://github.com/mne-tools/mne-python/blob/c9752455b79de10fac672d3ed25911f2618f71cc/mne/viz/backends/_pyvista.py#L1128-L1141

drammock, 2021-08-06:
> Is it an old version of mne?

it's in the windows 3.7 conda build, but it installs current `main`

closed 2021-09-20 by drammock