mne-tools/mne-python #9654: Inconsistent environment variable name for Sleep Physionet dataset

hubertjb, 2021-08-06:
The docstring of `mne.datasets.sleep_physionet.age.fetch_data` states the environment variable pointing to the Sleep Physionet dataset should be "MNE_DATASETS_PHYSIONET_SLEEP_PATH": https://github.com/mne-tools/mne-python/blob/855edbf48835bcee80e773eb7047d0a4caa14a1c/mne/datasets/sleep_physionet/age.py#L47

However the code looks instead for "PHYSIONET_SLEEP_PATH": https://github.com/mne-tools/mne-python/blob/855edbf48835bcee80e773eb7047d0a4caa14a1c/mne/datasets/sleep_physionet/_utils.py#L77

Given that other datasets, e.g. `eegbci` use the format "MNE_DATASETS_..._PATH": https://github.com/mne-tools/mne-python/blob/855edbf48835bcee80e773eb7047d0a4caa14a1c/mne/datasets/eegbci/eegbci.py#L62 I suppose the code should be changed to what the docstring states?

I just wanted to double check that is correct before submitting a PR. Thanks!

agramfort, 2021-08-07:
I think this dataset is not perfect when dealing with config. Besides the inconsistency of naming the dataset behaves properly? You can save the dataset where you want and it will update the mne config file in your home?
If so then I think it’s better just to update the doc string as otherwise we will break some code…

sappelhoff, 2021-08-08:
+1 with alex to just change the docstr if we want a quick fix.

For a good fix (more work, and some breakage), I think we should add a `data_path` function to the sleep_physionet dataset ... like we have for all (?) other datasets, and then make the naming consistent as well.

hubertjb, 2021-08-08:
@agramfort I just tested it and actually the `mne-config.json` file was **not** updated when I fetched data and provided a specific path. When I called it again without a path the data was downloaded again and saved under `~/mne_data`. Does that change what you'd recommend doing?

@sappelhoff So that would entail making a more complete version of the existing `_data_path` function? https://github.com/mne-tools/mne-python/blob/855edbf48835bcee80e773eb7047d0a4caa14a1c/mne/datasets/sleep_physionet/_utils.py#L44

agramfort, 2021-08-08:
> @agramfort I just tested it and actually the mne-config.json file was not updated when I fetched data and provided a specific path. When I called it again without a path the data was downloaded again and saved under ~/mne_data. Does that change what you'd recommend doing?

Yes I had noticed this in the past. The fetcher doesn’t behave like others. 

So maybe we will not break backward compat?
> @sappelhoff So that would entail making a more complete version of the existing _data_path function? https://github.com/mne-tools/mne-python/blob/855edbf48835bcee80e773eb7047d0a4caa14a1c/mne/datasets/sleep_physionet/_utils.py#L44
> 

Yes but I fear it’s quiet some work. Not sure how high priority this is…

drammock, 2021-10-12:
closed by #9657

closed 2021-10-12 by drammock