mne-tools/mne-python #9659: support Laplacian referencing for EEG

drammock, 2021-08-11:
#### Describe the new feature or enhancement

Add support for Laplacian (sometimes called local average) referencing for EEG.

Here is one reference: https://doi.org/10.1016/j.ijpsycho.2015.04.023 though it's part of a special issue on current source density analysis so there are likely other papers worth consulting before implementing.

#### Describe your proposed implementation

as a new option `mne.set_eeg_reference(ref_channels='local')`  The function would probably also need a new param for the distance threshold.

#### Describe possible alternatives

could make a dedicated function instead.  `set_eeg_reference` is already overloaded to handle REST referencing (with extra param `forward`) so I don't see the harm in extending it to Laplacian as well.

#### Additional comments

Requested on the forum https://mne.discourse.group/t/laplacian-or-local-average-reference-in-mne/3415

cross-ref to #8481 where Laplacian referencing was mentioned for sEEG, although I think the implementation for that would likely be fairly different, given the shaft-based vs scalp-based electrodes.

paulokanda, 2021-08-12:
More references:

Yao, D., Qin, Y., Hu, S., Dong, L., Vega, M. L. B., & Sosa, P. A. V. (2019). Which reference should we use for EEG and ERP practice?. Brain topography, 32(4), 530-549.

Abbass, H. A. (2014, November). Calibrating independent component analysis with Laplacian reference for real-time EEG artifact removal. In International Conference on Neural Information Processing (pp. 68-75). Springer, Cham.

Kayser, J., & Tenke, C. E. (2015). Issues and considerations for using the scalp surface Laplacian in EEG/ERP research: A tutorial review. International Journal of Psychophysiology, 97(3), 189-209.

Carvalhaes, C., & de Barros, J. A. (2015). The surface Laplacian technique in EEG: Theory and methods. International Journal of Psychophysiology, 97(3), 174-188.

Trujillo, L. T., Stanfield, C. T., & Vela, R. D. (2017). The effect of electroencephalogram (EEG) reference choice on information-theoretic measures of the complexity and integration of EEG signals. Frontiers in neuroscience, 11, 425.

Bringas Vega, M. L., Nunez, P., Riera, J., Zhang, R., & Valdes-Sosa, P. A. (2019). Through a glass, darkly: The influence of the EEG reference on inference about brain function and disorders. Frontiers in neuroscience, 13, 1341.

Levin, A. R., Varcin, K. J., O’Leary, H. M., Tager-Flusberg, H., & Nelson, C. A. (2017). EEG power at 3 months in infants at high familial risk for autism. Journal of Neurodevelopmental Disorders, 9(1), 1-13.

drammock, 2021-08-14:
ha!  I didn't know that CSD and surface laplacian were the same thing.  @paulokanda it looks like you can use `mne.preprocessing.compute_current_source_density(raw)`.  I'll close this, but if I'm mistaken and this doesn't meet your needs please respond here and we can re-open the issue.

closed 2021-08-14 by drammock

DominiqueMakowski, 2022-06-03:
If `mne.preprocessing.compute_current_source_density(raw)` is equivalent to what is described in the literature as ***rereference using a Laplacian (LAP) “reference-free” transformation*** (e.g., https://www.frontiersin.org/articles/10.3389/fnins.2017.00425/full), then it would be useful for visibility to have this as an explicit option in `mne.set_eeg_reference()` (and then explain in the docstring that it basically just runs the CSD transfo)

Just to clarify, does that mean that we don't need to first run `raw.set_eeg_reference()` before running `mne.preprocessing.compute_current_source_density(raw)`? Or should we still re-reference using for instance an average ref?

agramfort, 2022-06-03:
I would say try to see if it has any impact to average reference first.
Just by looking at the outputs.

larsoner, 2022-06-03:
IIRC the average ref is done within the Laplacian/CSD code internally, so it shouldn't matter whether or not you apply one first. But feel free to check in code or by example

>  then it would be useful for visibility to have this as an explicit option in mne.set_eeg_reference()

Since it changes the channel type in MNE it does a bit more than this function/method typically does. I'd rather leave it as a separate function, and just mention it somewhere in the `Notes` and/or `See Also`. Would you be up for updating the docstring of `set_eeg_reference` @DominiqueMakowski to reference the CSD function in some way that makes it more discoverable?