mne-tools/mne-python #9673: mismatch between automated coreg and GUI

drammock, 2021-08-18:
I tested out the new automated coreg today, and got different results from the old GUI:

#### GUI
[image]

I got this by setting fiducial point matching to 'mri fiducials' in the fit settings, then clicking "fit icp" under trans/rot (no scaling). 

#### automated + plot_alignment
[image]

code that led to the plot_alignment figure:

```py
raw_fname = os.path.join(data_root, subject, 'raw_fif', f'{subject}_raw.fif')
info = mne.io.read_info(raw_fname)
coreg = mne.coreg.Coregistration(
    info, subject=surrogate, subjects_dir=subjects_dir)
coreg.set_scale_mode('3-axis')
coreg._icp_fid_match = 'matched'
coreg.fit_icp(n_iterations=30)
mne.viz.plot_alignment(
    info, coreg.trans, subject=surrogate, subjects_dir=subjects_dir,
    surfaces=dict(head=0.9), dig=True, mri_fiducials=True, meg=False)
```

larsoner, 2021-08-18:
It just occurred to me that the weights for points might be different. HPI for example might default to 1. in the class but 0. in the GUI. You can see these in "Fitting options".

I'm hoping that's the problem, because if it's something else, it's going to be harder to figure out. I would suggest first checking if the alignment provided by just the fiducial match works. Once that does, do one ICP iteration, and make sure that matches.

I'm not 100% sure if the "fiducial distances" reported in the GUI are changed based on whether matched vs nearest is selected. IIRC in the Coregistration method `coreg.compute_dig_mri_distances` (where I think the fids are the first three values returned, maybe?) it is probably always be nearest...

larsoner, 2021-08-18:
... I would also suggest starting out with something easier like `sample` and not doing any scaling. So no scaling, fiducials fit, then one ICP iteration. Once these match to numerical precision, reprocess `sample` using the same steps, but with 3-axis enabled. Something like that that should help isolate the likely cause of the mismatch

PR #9690 (2021-08-23): MRG, MAINT: More complete coreg test and doc
Closes #9673

@drammock I realized the problem with the snippet:
```py
raw_fname = os.path.join(data_root, subject, 'raw_fif', f'{subject}_raw.fif')
info = mne.io.read_info(raw_fname)
coreg = mne.coreg.Coregistration(
    info, subject=surrogate, subjects_dir=subjects_dir)
coreg.set_scale_mode('3-axis')
coreg._icp_fid_match = 'matched'
coreg.fit_icp(n_iterations=30)
mne.viz.plot_alignment(
    info, coreg.trans, subject=surrogate, subjects_dir=subjects_dir,
    surfaces=dict(head=0.9), dig=True, mri_fiducials=True, meg=False)
```
is that the `plot_a [... cut]
closed 2021-08-23 by larsoner