mne-tools/mne-python #9679: mne.create_info() not creating a new instance (missing sfreq)

mpaccione, 2021-08-20:
#### Describe the bug
Using mne.create_info does not work when passing in other mne info to create a new info instance. 
The context is for EEG compression and saving new compressed versions.

#### Steps to reproduce
1. Read a file 
```
 original_raw = mne.io.read_raw(original_file_path, preload=True)
 info = original_raw.info
```
2. Print info and make sure that your info is correct
```
 print(info)
 print(info['sfreq'])
```
```
<Info | 7 non-empty values
 bads: []
 ch_names: EEG Fpz-Cz, EEG Pz-Oz, EOG horizontal, Resp oro-nasal, EMG ...
 chs: 7 EEG
 custom_ref_applied: False
 highpass: 0.5 Hz
 lowpass: 100.0 Hz
 meas_date: 1989-04-25 14:50:00 UTC
 nchan: 7
 projs: []
 sfreq: 100.0 Hz
>
100.0
```
3. Create new mne info instance to be used
```
new_info = mne.create_info({
        "bads": info['bads'],
        "ch_names": info['ch_names'],
        "chs": info['chs'],
        "custom_ref_applied": info['custom_ref_applied'],
        "highpass": info['highpass'],
        "lowpass": info['lowpass'],
        "meas_date": info['meas_date'],
        "nchan": info['nchan'],
        "projs": info['projs'],
        "sfreq": info['sfreq']
    })
```

#### Expected results
Info instance is created and then to be used in the following code to save a compressed version

```
   new_raw = mne.io.RawArray(channels, new_info)
    f = None

    if compress:
        f = lzma.LZMAFile(new_file_path, mode="wb")
    else:
        f = open(new_file_path, "a")

    f.write(new_raw)
    f.close()    
```

#### Actual results
ERROR: TypeError: create_info() missing 1 required positional argument: 'sfreq'
[image]

#### Additional information
```
Platform:      Linux-5.8.0-63-generic-x86_64-with-glibc2.32
Python:        3.8.10 (default, Jun  2 2021, 10:49:15)  [GCC 10.3.0]
Executable:    /usr/bin/python3
CPU:           x86_64: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.22.1
numpy:         1.20.2 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.4.1 {backend=Qt5Agg}

sklearn:       0.24.1
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.4
mayavi:        Not found
pyvista:       0.29.1 {pyvistaqt=0.3.0, OpenGL 4.5.0 NVIDIA 390.144 via GeForce GTX 970/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.0
```

agramfort, 2021-08-20:
look at the signature of create Info:

https://mne.tools/stable/generated/mne.create_info.html

it does not take the dict you pass as input.

drammock, 2021-08-20:
I think also you can save yourself some typing by doing `new_info = info.copy()`

mpaccione, 2021-08-20:
Ah gotcha info.copy is definitely the way to go in this situation! Thank you.

closed 2021-08-20 by mpaccione