mne-tools/mne-python #9683: Colorbar does not plot for viz.topoplot

moonj94, 2021-08-20:
I am trying to visualize a topography with a colorbar, but I cannot for the life of me get the colorbar to show. 

Here is my code:

```
fig,ax=plt.subplots(1,1,figsize=(6,6),dpi=1200)
clim = dict(kind='value', lims=[4, 8, 12])

im,cm=mne.viz.plot_topomap(dist[0,:],raw_concat.info,contours=5,axes=ax,cmap='viridis',vmin=np.min(dist),vmax=np.max(dist))
ax_x_start = 0.95
ax_x_width = 0.04
ax_y_start = 0.1
ax_y_height = 0.9
cbar_ax = fig.add_axes([ax_x_start, ax_y_start, ax_x_width, ax_y_height])
# clb = plt.colorbar(cm, cax=cbar_ax,ax=ax)
unit_label='MDS distance'
# plt.colorbar(cm)
clb = fig.colorbar(cm, cax=cbar_ax,ax=ax)
clb.ax.set_title(unit_label,fontsize=14) # title on top of colorbar
plt.show()
```

I have also tried (following matplotlib example):
```
fig,ax=plt.subplots(1,1,figsize=(6,6),dpi=1200)
clim = dict(kind='value', lims=[4, 8, 12])

im,cm=mne.viz.plot_topomap(dist[0,:],raw_concat.info,contours=5,axes=ax,cmap='viridis',vmin=np.min(dist),vmax=np.max(dist))

divider = make_axes_locatable(ax)
cax = divider.append_axes("right", size="5%", pad=0.05)
plt.colorbar(im,cax=cax,ax=ax)
```

Your help is very much appreciated!

agramfort, 2021-08-21:
```
from pathlib import Path
import numpy as np
import matplotlib.pyplot as plt
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)
raw.pick('eeg')

fig, (ax, cbar_ax) = plt.subplots(1, 2, figsize=(6, 4), gridspec_kw=dict(width_ratios=[6, 1]))
clim = dict(kind="value", lims=[4, 8, 12])

dist = 10e-6 * np.random.randn(1, len(raw.ch_names))

im, cm = mne.viz.plot_topomap(
    dist[0, :],
    raw.info,
    contours=5,
    axes=ax,
    cmap="viridis",
    vmin=np.min(dist),
    vmax=np.max(dist),
)
unit_label = "MDS distance"
clb = fig.colorbar(im, cax=cbar_ax, ax=ax)
clb.ax.set_title(unit_label, fontsize=14)  # title on top of colorbar
plt.show()
```
this should work.

close the issue if so.

moonj94, 2021-08-23:
Hi Alexandre, thanks for the help. Your code helped me put the colorbar axes in the same figure, but I am not able to plot the colors themselves on the axes. How can I achieve this?

agramfort, 2021-08-23:
sorry I don't get what you mean

>

moonj94, 2021-08-23:
When I use the code you wrote, this is the product:
[image]

I can't get any colors on the colorbar itself. and the colorbar title also does not show

agramfort, 2021-08-23:
my code gives me this:

[image]

moonj94, 2021-08-23:
thats odd, how can I troubleshoot this issue? Do you think it could be an issue with Spyder? I should say that I am using spyder 4.2.1. I can't seem to download any version higher than that because of dependency conflicts

drammock, 2021-08-23:
> Do you think it could be an issue with Spyder?

try it in a system console with Python or IPython; if it works there and fails from within Spyder, then it's almost surely an issue with Spyder

Other things to try (in order):

1. see whether running `import matplotlib.pyplot as plt; plt.ion()` first fixes it; if so, one of the earlier lines was blocking execution of the later lines, so they weren't taking effect
2. *in the terminal where the plot is failing*, run `mne.sys_info()` and report the results here.

agramfort, 2021-08-23:
can you try directly from python or a notebook and not from spyder?

>

moonj94, 2021-08-23:
@drammock I tried plt.ion and it didn't make a difference.

I also tried updating every package including spyder and matplotlib and the issue still persists.

@agramfort I tried running the script in python3 terminal/bash and i still do not get any colors.

drammock, 2021-08-23:
> 2. _in the terminal where the plot is failing_, run `mne.sys_info()` and report the results here.

moonj94, 2021-08-23:
@drammock I updated my previous comment

drammock, 2021-08-23:
I can reproduce your error.  But I can also fix it by adding `plt.ion()` to @agramfort's code example (after the `import` statements).  Are you *sure* you tried including `plt.ion()` *before* the figure was created?

moonj94, 2021-08-23:
@drammock I tried putting plt.ion() right after importing and before the figure was generated. This is the script:

```
import scipy.io
import os
import numpy as np
# from mapping.barplot_annotate_brackets import barplot_annotate_brackets
import mne
from mne import io
import matplotlib.pyplot as plt
plt.ion()
mat=scipy.io.loadmat('/Volumes/tempdata/Study2/MDS/dist.mat')
dist=mat['dist']
pval=mat['pval']
#  topoplot

pvalpass=pval>0.05
dist[pvalpass]=0
# from mpl_toolkits.axes_grid1 import make_axes_locatable

data_path2='/Volumes/data2/Study2/Raw/P98/S02'
# data_path2='/Volumes/data2/Study1/Raw/P{0:02d}'.format(ps[participant])+'/S02'
for file in os.listdir(data_path2):
    if file.endswith('.vhdr'):
        data_path2=os.path.join(data_path2,file)
    # for file in os.listdir(data_path2):
    #     if file.endswith('.vhdr'):
    #         data_path2=os.path.join(data_path2,file)

raw_concat=io.read_raw_brainvision(vhdr_fname=data_path2,preload=True)   
# actual topoplot

# dist = 10e-6 * np.random.randn(1, len(raw.ch_names))
fig, (ax, cbar_ax) = plt.subplots(1, 2, figsize=(6, 4), gridspec_kw=dict(width_ratios=[6, 1]))
clim = dict(kind="value", lims=[4, 8, 12])
plt.ion()
im, cm = mne.viz.plot_topomap(
    dist[0, :],
    raw_concat.info,
    contours=5,
    axes=ax,
    cmap="viridis",
    vmin=np.min(dist),
    vmax=np.max(dist),
)
unit_label = "MDS distance"
# plt.colorbar()
clb = fig.colorbar(cm, cax=cbar_ax, ax=ax)
clb.ax.set_title(unit_label, fontsize=14)  # title on top of colorbar
plt.show(block=True)
```

This is the figure I keep getting:
[image]

drammock, 2021-08-23:
It's difficult to debug when your script references paths and data objects that I can't access.  Here is @agramfort's script again, with `plt.ion()` added in the latest possible place (just to emphasize where it needs to go).  Can you please test **this script** instead of the one using your own data, and tell us if it works?

```
from pathlib import Path
import numpy as np
import matplotlib.pyplot as plt
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)
raw.pick('eeg')
plt.ion()  # <<<<<< this needs to happen **before** you create the figure!!
fig, (ax, cbar_ax) = plt.subplots(1, 2, figsize=(6, 4), gridspec_kw=dict(width_ratios=[6, 1]))
clim = dict(kind="value", lims=[4, 8, 12])

dist = 10e-6 * np.random.randn(1, len(raw.ch_names))

im, cm = mne.viz.plot_topomap(
    dist[0, :],
    raw.info,
    contours=5,
    axes=ax,
    cmap="viridis",
    vmin=np.min(dist),
    vmax=np.max(dist),
)
unit_label = "MDS distance"
clb = fig.colorbar(im, cax=cbar_ax, ax=ax)
clb.ax.set_title(unit_label, fontsize=14)  # title on top of colorbar
plt.show()
```

moonj94, 2021-08-23:
I am still getting the same result, with plt.ion placed before plt.subplots

moonj94, 2021-08-23:
I tried fig.colorbar (im, cax=cbar_ax,ax=ax) in bash and it seems to have worked (used cm previously). Not sure why it is not working on spyder, however.

Thanks @agramfort @drammock

closed 2021-08-23 by agramfort

drammock, 2021-08-23:
FYI, I just had a chance to test in spyder.  If you add `show=False` to the `plot_topomap` call, then it works with the inline backend too.