mne-tools/mne-python #9703: suspicious behavior  in pick function in channels (UpdateChannelsMixin.pick)

evgenygoldstein, 2021-08-29:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should ask for help on the
[MNE Forum](https://mne.discourse.group).

Our GitHub issue tracker is only used to report bugs and suggest improvements
to MNE-Python. For any other questions, please use the forum.
Usage questions that are posted as GitHub issues are usually closed without
being answered. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
please **delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block (see
[the GitHub guide to markdown](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown)
for help with issue formatting). Alternatively, you can put your MWE in a
[public gist](https://gist.github.com) and link to it in this issue.

#### Describe the bug
when working with raw data of eeg and trying to filter the channels of raw data class by using
"pick" function the results might be unexpected, the returned raw file could contain less channels then you would expect.

#### Steps to reproduce
import mne
from mne.datasets import sample

print(__doc__)

data_path = sample.data_path()

raw = mne.io.read_raw_fif(data_path + '/MEG/sample/sample_audvis_raw.fif')
print(raw.ch_names) # a very long list of channels names 'MEG 0113', 'MEG 0112',  'MEG 0111', 'MEG 0122', 'MEG 0123', 'MEG 0121'.......
raw = raw.pick(['MEG 0113', 'MEG 0112', 'Meg 0111', 'MEG 0111', 'MEG 0122', 'MEG 0123', 'MEG 0121', 'MEG 0132', 'MEG 0133', 'MEG 0131', 'MEG 0143', 'MEG 0142', 'MEG 0141', 'MEG 0213', 'MEG 0212', 'MEG 0211', 'MEG 0222', 'MEG 0223', 'MEG 0221', 'MEG 0232', 'MEG 0233', 'MEG 0231', 'MEG 0243', 'MEG 0242', 'MEG 0241', 'MEG 0313', 'MEG 0312', 'MEG 0311', 'MEG 0322', 'MEG 0323', 'MEG 0321', 'MEG 0333', 'MEG 0332', 'MEG 0331', 'MEG 0343', 'MEG 0342', 'MEG 0341', 'MEG 0413', 'MEG 0412', 'MEG 0411', 'MEG 0422', 'MEG 0423', 'MEG 0421', 'MEG 0432', 'MEG 0433', 'MEG 0431', 'MEG 0443', 'MEG 0442', 'MEG 0441', 'MEG 0513', 'MEG 0512', 'MEG 0511', 'MEG 0523', 'MEG 0522', 'MEG 0521', 'MEG 0532', 'MEG 0533', 'MEG 0531', 'MEG 0542', 'MEG 0543', 'MEG 0541', 'MEG 0613', 'MEG 0612', 'MEG 0611', 'MEG 0622', 'MEG 0623', 'MEG 0621', 'MEG 0633', 'MEG 0632'])
print(raw.ch_names) # list of only two channels

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
['MEG 0113', 'MEG 0112', 'MEG 0111', 'MEG 0122', 'MEG 0123', 'MEG 0121', 'MEG 0132', 'MEG 0133', 'MEG 0131', 'MEG 0143', 'MEG 0142', 'MEG 0141', 'MEG 0213', 'MEG 0212', 'MEG 0211', 'MEG 0222', 'MEG 0223', 'MEG 0221', 'MEG 0232', 'MEG 0233', 'MEG 0231', 'MEG 0243', 'MEG 0242', 'MEG 0241', 'MEG 0313', 'MEG 0312', 'MEG 0311', 'MEG 0322', 'MEG 0323', 'MEG 0321', 'MEG 0333', 'MEG 0332', 'MEG 0331', 'MEG 0343', 'MEG 0342', 'MEG 0341', 'MEG 0413', 'MEG 0412', 'MEG 0411', 'MEG 0422', 'MEG 0423', 'MEG 0421', 'MEG 0432', 'MEG 0433', 'MEG 0431', 'MEG 0443', 'MEG 0442', 'MEG 0441', 'MEG 0513', 'MEG 0512', 'MEG 0511', 'MEG 0523', 'MEG 0522', 'MEG 0521', 'MEG 0532', 'MEG 0533', 'MEG 0531', 'MEG 0542', 'MEG 0543', 'MEG 0541', 'MEG 0613', 'MEG 0612', 'MEG 0611', 'MEG 0622', 'MEG 0623', 'MEG 0621', 'MEG 0633', 'MEG 0632']

#### Actual results
['MEG 0113', 'MEG 0112']

#### Additional information
the reason is that a loop in _picks_str_to_idx, pick.py:1111
(_picks_str_to_idx, pick.py:1111
_picks_to_idx, pick.py:1045
pick, channels.py:838
<module>, mne_bug_report.py:10)

stops iteration the moment it encounter name of the channel which is not in the raw instance original list of names
    for pick in picks:
        try:
            picks_name.append(info['ch_names'].index(pick))
        except ValueError:
            bad_name = pick
            break

agramfort, 2021-08-29:
you have written 

`'Meg 0111', 'MEG 0111'`

if you remove the `'Meg 0111'` it works fine although I agree that you should have a clear error message...

evgenygoldstein, 2021-08-29:
the problem from my point of view, is that list of channel names provided as a parameter to the pick function have to be a subset of Raw instance channel names, otherwise the function changes the instance in unexpected fashion (since it stops selecting channels at the first unincluded channel name). That fact wasn't clear  from the function description, also without some kind of error message it is easy to miss. It will take a small change to the iteration to keep checking the list of names
after encountering unincluded channel name.

agramfort, 2021-08-30:
I agree it's not the behavior we should be expecting.

can you send us a PR to fix this?

>

evgenygoldstein, 2021-09-01:
done

agramfort, 2021-09-02:
@evgenygoldstein you opened a PR?

commit 6ab3c551 (2021-09-07): FIX: #9703 suspicious behavior in pick function (#9708)

* FIX: pick function behaves unexpectedly

closes  #9703
closed 2021-09-07 by agramfort