mne-tools/mne-python #9704: Improving EDF reading in line with specification

adam2392, 2021-08-30:
#### Describe the new feature or enhancement
In https://github.com/mne-tools/mne-python/pull/9694#issuecomment-908135503, I realized that the EDF specification can "encode" the channel type. One can have a label string be:

`EEG Fz`, implying that it is the Fz channel of `eeg` type. However, it's "optional".

For exporting EDF data, I would like to augment the export to encode the channel type, such that if you have channels:

`EOG1, ECG1, A1, A2, A3, Fz, Cz`, then they would be written to EDF like:

```
EOG EOG1
ECG ECG1
sEEG A1
sEEG A2
sEEG A3
EEG Fz
EEG Cz
```

#### Describe your proposed implementation
First, I would modify these lines: https://github.com/mne-tools/mne-python/blob/22cd7747529db7895345917c4d3884a482ed8ab3/mne/io/edf/edf.py#L387-L416

to allow for backwards-compatible behavior that also then sees if there is the "channel type" encoded in the label string.

#### Additional comments
That way I can modify the string of PRs: 
- https://github.com/mne-tools/mne-python/pull/9694
- #9643

adam2392, 2021-08-30:
If core-devs agree this is desired behavior, then I can add a PR.

larsoner, 2021-08-30:
Has anyone ever encountered such files in the wild? If not, then I wouldn't add support for it until we hit a use case where it's actually needed

adam2392, 2021-08-30:
All my files actually have the `EEG` (i.e. channel type) prefix in the channel label. I can add a dummy file to misc/testing, but I think maybe one of the EDF files have it too. I can double check.

[image]

larsoner, 2021-08-30:
Okay sounds justified enough to me! Bonus points if some testing data are aleady formatted that way but even if not I'm happy with you having some files.

For my edification, what system/conversion routine produced those files with the channel type encoded?

adam2392, 2021-08-30:
I believe the recording system at this center used Nihon Kohden. However, it might've done the conversion through Persyst instead.

My suspicion is that not "all files" have this behavior as the "string channel label" in EDF is not strictly required to have the channel type. It's just a pattern in the string that is available if need be. I.e. I've seen some files from Nihon Kohden (or Natus) converted that don't have it. :( 

Therefore, I don't think the change will be that bad since it's really just adding an if/else clause capturing the "space" in the string label if it detects there are.

adam2392, 2021-08-30:
Added PR to mne-testing-data: https://github.com/mne-tools/mne-testing-data/pull/88

larsoner, 2021-08-30:
+1 for adding a `ch_type` mapping for this during reading (and exporting) in MNE-Python from me then. Hopefully we support all channel types EDF does, but any that we don't we should probably map to EEG (?) and issue a warning.

adam2392, 2021-08-30:
Yeah it seems like they map to EEG by default currently, so I'm okay with just leaving that as is. 

Do you think a warning is necessary if that's current behavior? I suppose for a newbie to MNE, having channel "DC01" map to EEG because it doesn't have a channel type wouldn't be expected.

larsoner, 2021-08-30:
Indeed I would make a list of all the channel names that had a type that we didn't know and say we mapped them to EEG at the end, something like:
```
bad_map = dict()
for ci in range(channels):
    if this_ch_type not in ch_type_dict:
        bad_map[ch_name] = this_ch_type
        this_ch_type = 'eeg'
if len(bad_map):
    bad_map = '\n'.join(f'{ch_name}: {ch_type}' for ch_name, ch_type in bad_map.items()'
    warn(f'Found the following unknown channel type mapping(s), setting the channel type to EEG:\n{bad_map})
```

cbrnr, 2021-08-30:
I have never used EDF files with this naming convention. Can you check the names in https://physionet.org/content/eegmmidb/1.0.0/? I'm OK with deducing the channel type if the names follow this scheme, but I'm -1 on modifying existing labels on export.

adam2392, 2021-08-30:
Re reading it in: Not all files will have it. I showed a screenshot of a file that does. Many files though may just have the channel name. This is just a consequence of different manufacturing systems exporting EDF files differently.

Re exporting: The [standard](https://edfplus.info/specs/edftexts.html#label) has this built in, so it wouldn't hurt to have it export if the channel type information is there. One would lose roundtrip ability with 'misc', and 'bio' channels perhaps, but everything else is preserved as intended i.e. the Voltage channels EEG, ECOG, SEEG, etc.

cbrnr, 2021-08-30:
> Re exporting: The standard has this built in, so it wouldn't hurt to have it export if the channel type information is there. One would lose roundtrip ability with 'misc', and 'bio' channels perhaps, but everything else is preserved as intended i.e. the Voltage channels EEG, ECOG, SEEG, etc.

Then at least let's have a parameter that controls this behavior.

adam2392, 2021-08-30:
So if you have the following channels:
```
{
   A1 = 'eeg',
   A2 = 'eeg',
   ECG1 = 'ecg',
   EOG1 = 'eog',
   LMT1 = 'seeg'
}
```

Then a parameter would either let their labels be written as the following in the EDF file?
```
A1, A2, ECG1, EOG1, LMT1
```

or

```
EEG A1, EEG A2, ECG ECG1, EOG EOG1, SEEG LMT1
```

Is there any reason to want the labels be the first way, rather than the second? Mistakes by user?

adam2392, 2021-08-30:
Well I suppose one would want to toggle it yes when exporting from a FIFF file where the original names are:

```
EEG A1, EEG A2, ECG ECG1, EOG EOG1, SEEG LMT1
```

One wouldn't want to accidentally add an extra channel type string in front of each of those.

PR #9694 (2021-09-15): [MRG, BUG] Fixed bug with EDF channel types and minor bug where other types of channels were not written and bug where annotations were not written to correct scale
#### Reference issue
Follow up fix from #9643 where not all the relevant "Voltage/uVoltage" channels were added.

In addition, it will allow channel types to be written to EDF files, in accordance with: https://github.com/mne-tools/mne-python/pull/9694#issuecomment-908135503

Closes: #9704 
Closes: #9730

#### What does this implement/fix?
- Adds corresponding unit test that makes sure all voltage channels are covered in exporting  [... cut]
closed 2021-09-15 by agramfort