mne-tools/mne-python #9720: Is the stc.plot hemi parameter broken?

rob-luke, 2021-09-07:
#### Describe the bug
For my own code plotting source estimates only the right hemisphere is no showing, even if I set `hemi="both`

#### Steps to reproduce
I think this may also be evident in the examples like

https://mne.tools/dev/auto_tutorials/inverse/40_mne_fixed_free.html#fixed-orientation

https://mne.tools/dev/auto_tutorials/stats-source-space/20_cluster_1samp_spatiotemporal.html#visualize-the-clusters

#### Expected results

[image]

#### Actual results

[image]

#### Additional information

agramfort, 2021-09-07:
it seems so...

>

larsoner, 2021-09-07:
I can look into this today hopefully

closed 2021-09-07 by drammock

rob-luke, 2021-09-07:
Thanks, that was quick!

PR #9721 (2021-09-07): MRG, BUG: Fix hemi
Closes #9720

Fixes regression introduced by #9688 (specifically on [this line](https://github.com/mne-tools/mne-python/pull/9688/files#diff-4e3d2d919a60057d535ba733330d5b8338aa491d6392b3d1c4a1a3de7122e17eR1930)), adds a test that would have caught it, touches an example that is broken on `main`, and fixes a persistent `linkcheck` false alarm.