mne-tools/mne-python #9733: [Improvement to EDF exporting and reading] Implement reading/exporting MEG/Fnirs channel types to EDF files

adam2392, 2021-09-14:
@cbrnr wanted the ability to export MEG/Fnirs types to EDF files. In order to allow export, we should also be able to read these data channels back in to the best of our ability.

A PR should enable:
- exporting of these channel types to EDF 
- allow reading back in these channel types via `read_raw_edf`

Actually the challenge is more-so on reading in MEG/fnirs data correctly using `read_raw_edf`. I'm not sure how to read in this data correctly. Sure I can easily append "MEG M1" to indicate M1 is a MEG channel, or "FNIRS M2" to indicate M2 is a FNIRS channel, but reading the data back in to do a unit test on the round trip is non-trivial for me.

Is it possible to leave this PR as is and then table it for a future improvement from someone who wants to export MEG/Fnirs to EDF and is familiar with the reader of MEG/Fnirs enough to add that functionality in?

Reference LOC: https://github.com/mne-tools/mne-python/blob/b579d328cb67d6a0096ff5ba74e4e2e557e84065/mne/io/edf/edf.py#L388-L417

_Originally posted by @adam2392 in https://github.com/mne-tools/mne-python/pull/9694#discussion_r708340358_

larsoner, 2021-09-14:
I think for MEG at least the answer is "no" because too much information will be lost (sensor orientations, head<->MEG transformation, sensor coil types). For fNIRS they have their own standard format (SNRF) so I don't see the value added. I do see a clear drawback in that it's more code for us to maintain that might or might not get used and might or might not work all that well.

I'd rather not bother (and in the meantime close this issue as WONTFIX) until someone who really wants this figures out how to implement it properly, probably after discussing with the EDFLib folks first (if it hasn't happened yet) to make sure it's reasonable and that we aren't abusing the format.

cbrnr, 2021-09-14:
EDF is a format for "polygraphic recordings", it can store any kind of signal and it is not restricted to voltage units. I suggested that we allow exporting arbitrary channels, but we don't have to make it more complicated than necessary. For example, if we have a temperature channel (in degrees Celsius), we should still be able to export it. When reading such a channel, just assign  it a `misc` type.

closed 2021-09-15 by cbrnr