mne-tools/mne-python #9734: Getting atributeError when plotting sensitivity map

balandongiv, 2021-09-14:
When running the the tutorial:

https://mne.tools/stable/auto_tutorials/forward/35_eeg_no_mri.html

The compiler return an error

AttributeError: 'NoneType' object has no attribute '_widget'
  brain = eeg_map.plot(time_label='EEG sensitivity', subjects_dir=subjects_dir,
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/source_estimate.py", line 648, in plot
    brain = plot_source_estimates(
  File "<decorator-gen-158>", line 24, in plot_source_estimates
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/viz/_3d.py", line 1730, in plot_source_estimates
    return _plot_stc(
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/viz/_3d.py", line 1908, in _plot_stc
    brain.setup_time_viewer(time_viewer=time_viewer,
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/viz/_brain/_brain.py", line 686, in setup_time_viewer
    self._configure_playback()
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/viz/_brain/_brain.py", line 1154, in _configure_playback
    self._renderer._playback_initialize(
  File "/home/host/anaconda3/envs/prj/lib/python3.9/site-packages/mne/viz/backends/_notebook.py", line 243, in _playback_initialize
    slider = time_widget._widget
AttributeError: 'NoneType' object has no attribute '_widget'

Process finished with exit code 1

mscheltienne, 2021-09-30:
Running the tutorial section by section or all at once in Spyder worked for me on the latest 0.24 dev version on Windows.

balandongiv, 2021-10-01:
Same root cause as the issue raise at https://github.com/mne-tools/mne-python/issues/9788.

Suggestion made by eric indirectly solve the issue here. https://github.com/mne-tools/mne-python/issues/9788#issuecomment-929493736

closed 2021-10-01 by balandongiv