mne-tools/mne-python #9736: Adding an optional token to the dataset fetcher code to allow optional fetching from private repositories

adam2392, 2021-09-15:
#### Describe the new feature or enhancement
The dataset fetching code inside `mne/datasets/utils.py`, `mne/utils/fetching.py` are actually very general. I was hoping to leverage them without copy/pasting the code, so I can make use of upstream possible bug fixes / performance improvements (if they ever occur). 

However, in some cases, I would like to unit test against private data I have stored on Github, and they require an API token with the HTTP request. Eventually, then some of that data would be made private after say a publication, but it's then nice to build into a CI for myself for a private research project in the meantime. 

Is it possible to add an optional "token" into the dataset fetcher?

This would also enable MNE to leverage private repos. In addition, it would lessen the code dependency for anyone trying to implement a data fetcher without copying every single function from MNE.

#### Describe your proposed implementation
Add optional `token=None` kwarg to the following functions:

1. `_download`
2. `_fetch_file`
3. `_get_http`

Then one can easily add optional tokens in `_data_path`, depending on which dataset is being fetched.

#### Describe possible alternatives
This would also enable any "mne" package, like mne-bids/connectivity/etc. to leverage private Github repo data that might get passed in via GH actions?

If we further refactor things, so that `key`, `urls`, `archive_names`, `folder_origs`, `folder_names`, `md5_hashes` are passed into `_data_path`, rather then set inside `_data_path`, then to create a MNE-fetcher, one simply needs to define a `data_path` that then passes these to `_data_path`, and they have a fully functional: `mne_downstream.testing.data_path()`

rob-luke, 2021-09-15:
This is an interesting proposal @adam2392, I also have a private code repository for handling fetching my various datasets, so offloading some of that to MNE would be advantageous to me too. However, ...

> I would like to unit test against private data I have stored on Github

Maybe this is more a curiosity, but do you actually store your data on GitHub? Looking at the LFS storage on GH its $5 per 50 GB per month ($0.1/GB), as opposed to something like backblaze at $0.005/GB. Or are you just talking about small testing datasets that fall under the 1GB limit?

drammock, 2021-09-15:
Interesting idea. I'm not opposed in principle. Another option might be Pooch; not sure if they have support for API tokens already. 

About 9 months ago I worked on migrating MNE-Python to use Pooch so that we could get rid of a lot of this code, but I didn't finish it because it took them so long to merge my enhancement PR. I'm mentioning this so you can investigate whether (1) you want to work on that migration with me, or (2) a future switch to Pooch would be incompatible with what you want to do.

adam2392, 2021-09-15:
^ small testing datasets that fall under the 1GB limit, but we can't make public "yet". And yeah I basically have like 4 different version of the current MNE fetcher code, but they all modify maybe like... 10 LOC, so that suggested to me that this would be a valuable refactoring in MNE.

It seems nowadays as well that MNE is more and more of a "platform", since it "enables" analysis and testing related to MEG/EEG/iEEG and then offloads analysis and more niche stuff to other mne packages, like mne-connectivity, mne-bids, etc. Part of this enablement in my opinion is making data fetching easier for CI/unittesting.

> This is an interesting proposal @adam2392, I also have a private code repository for handling fetching my various datasets, so offloading some of that to MNE would be advantageous to me too. However, ...

:) I'm not alone

adam2392, 2021-09-15:
> Interesting idea. I'm not opposed in principle. Another option might be Pooch; not sure if they have support for API tokens already.

Ooo interesting. I don't know what Pooch is, but I'll go read about it rn.

> About 9 months ago I worked on migrating MNE-Python to use Pooch so that we could get rid of a lot of this code, but I didn't finish it because it took them so long to merge my enhancement PR. I'm [...]

Hmm can you link to the corresponding issues/PRs? If its within my abilities / doesn't take an enormous amount of time, I would definitely be down to work on this.

^ if it does take a lot of effort though, is there an opposition to the refactoring plan I suggested though? Or is that somewhat contradictory work?

adam2392, 2021-09-15:
Okay found your draft: https://github.com/mne-tools/mne-python/pull/8679

adam2392, 2021-09-15:
Also regarding Pooch having support for an API token: https://www.fatiando.org/pooch/latest/api/generated/pooch.HTTPDownloader.html#pooch.HTTPDownloader

looks like one can set up an HTTP downloader that passes in kwargs to `request.get`, which supports tokens in the request header.

adam2392, 2021-09-16:
Now that #9742 is almost done, what is your thoughts on generalizing the data fetching module with some documentation to easily allow any downstream package to create their own neural testing dataset (optionally in a private repo)?

I'm thinking an API would be a public version of the current `_data_path()`, and something like:

`fetch_dataset(checksums_file_path, archive_map, url_map, folder_name_map, config_map=None, **_data_path_kwargs)`

Then in the docstring, we can specify how one can create their own checksums file, and then just pass in some parameters and they leverage the MNE dataset fetcher with the same backbone. This can then be useful for say... mne-connectivity/mne-bids/etc. to store its own datasets, without having to make a PR to mne-python's datasets that would just increase the size there.

cc: @larsoner @agramfort

agramfort, 2021-09-17:
let's discuss during office hours today

>

adam2392, 2021-09-17:
Copying over here for next 2 PRs:

1. restructure dataset info as nested dict of dicts; incorporate MD5s into python and change pooch to temporarily write the registry file (separate PR) and whatever else is necessary to make it easier to add new built-in datasets
2. add generic dataset downloader function, that allows e.g., API keys (separate PR). This should probably look like a data_path() function that takes in URL, archive name, hash, etc, and does everything that one of our built-in data_path() functions does (except for checking/setting config keys)

Ref link: https://github.com/mne-tools/mne-python/pull/9742#issuecomment-921932044

PR #9763 (2021-09-25): [MRG, MAINT] Fetching Datasets Public API "fetch_dataset" used by all internal MNE datasets now
#### Reference issue
Closes #9736 
Closes #9774

#### What does this implement/fix?
This adds the public function `mne.datasets.fetch_dataset` and refactors the existing MNE datasets to use that function.

This also refactors `_data_path` function to be more generic.
closed 2021-09-25 by larsoner