mne-tools/mne-python #9737: [BUG] ECoG tutorial plot sensors plots are out of scale

alexrockhill, 2021-09-15:
https://mne.tools/dev/auto_tutorials/clinical/30_ecog.html

In this tutorial, it looks like `brain.plot_sensors` is off. I think this may have to do with the fix @GuillaumeFavelier did?

agramfort, 2021-09-15:
which change from @GuillaumeFavelier do you have in mind?

>
maybe a pyvista update?

larsoner, 2021-09-15:
The `surfs` dict gets created but they never plotted. These lines were deleted but never replaced:

https://github.com/mne-tools/mne-python/pull/9714/files#diff-51ccb8d036c868f218d2fe19dbf7ee33fb7f9ceb5fde22ca2cda6192f0061fd0L719-L726

Clearly our unit tests are deficient... I can look into a fix.

closed 2021-09-15 by larsoner

alexrockhill, 2021-09-15:
It still looks like it's off the screen and there's no brain in the background in the ECoG tutorial

larsoner, 2021-09-15:
> It still looks like it's off the screen and there's no brain in the background in the ECoG tutorial

Where? The #9741 PR build artifact is here, it looks okay to me:

https://34197-1301584-gh.circle-artifacts.com/0/dev/auto_tutorials/clinical/30_ecog.html

https://mne.tools/dev has not updated, it takes ~2h after merge and this only went in 20 minutes ago.

alexrockhill, 2021-09-15:
You're right, I was just impatient sorry!

PR #9741 (2021-09-15): MRG, BUG: Fix surface plot regression
Closes #9737

Within-cycle bugfix so no `latest.inc` update required