mne-tools/mne-python #9744: Plot sensors is broken, not showing sensors in rendered docs.

rob-luke, 2021-09-15:
#### Describe the bug
The plot sensors function is not plotting sensors on the fNIRS tutorial (I haven't checked other tutorials, may be general too).

#### Steps to reproduce / Actual results

See dev docs https://mne.tools/dev/auto_tutorials/preprocessing/70_fnirs_processing.html#view-location-of-sensors-over-brain-surface

#### Expected results
See stable docs https://mne.tools/stable/auto_tutorials/preprocessing/70_fnirs_processing.html#sphx-glr-auto-tutorials-preprocessing-70-fnirs-processing-py

#### Additional information
I have seen lots of activity with the 3D brain visualisation lately, so I am not sure who to ping, so everyone related gets a ping! @agramfort @larsoner @GuillaumeFavelier @alexrockhill

larsoner, 2021-09-15:
Can you try locally on latest `main`? The fixes from #9741 have not landed in `mne.tools` yet (should be in 2h hopefully now that #9743 is in)

larsoner, 2021-09-15:
... if not, then this might be due to #9737 or something similar

rob-luke, 2021-09-15:
> should be in 2h 

I may have some programming time this afternoon, so more than 2h from now, so I will report back later today if this gets fixed by the new PRs :)

rob-luke, 2021-09-16:
Just checked, it's not fixed.

PR #9750 (2021-09-16): MRG, ENH: Add cortex, fix sensors
Closes #9744

1. Fixes bug introduced in #9585 (cc @alexrockhill) where fNIRS and projected EEG sensors were not scaled properly based on `units`
2. Fixes a bug where we iterated over some views twice (in the case of `hemi='both'`) when we should have just done it once -- this is fixed by iterating over `'vol'` as the hemi
3. Adds support for custom `cortex` color mappings in `mne.viz.Brain`, which is something I've wanted for a while and was necessary to restore this image from [stable](https://mne.tools/stable/auto_tutorials/preprocessing/70_fnirs_processing [... cut]
closed 2021-09-16 by larsoner