mne-tools/mne-python #9747: make_fixed_length_epochs() should allow for custom specification of the event ID

hoechenberger, 2021-09-16:
Currently, `make_fixed_length_epochs()` assigns all events the ID `1`.

`make_fixed_length_events()`, on the other hand, already supports user-defined IDs.

It would be good to add a parameter to `make_fixed_length_epochs()`, allowing users to pass a custom event ID.

This was brought up on the forum, https://mne.discourse.group/t/fixed-epochs-and-fixed-events/3702

mscheltienne, 2021-09-24:
I wanted to quickly fix this as I might be using it soon as well but it's not as straightforward as I thought.

If we look at the discourse post, the user has a similar RAW to mine: multiple events of different lengths. In my case, I have an alternation of regulation phase (16s) and of non-regulation phase (8s). The beginning of each phase is marked with an event, 5 for regulation and 6 for non-regulation. The display on MNE browser with each event marked as an annotation is:

[image]

<details open>
<summary>MNE event array</summary>
<be>

```
mne.find_events(raw, 'TRIGGER')
array([[  5528,      0,      6],     # first non-regulation phase is longer, don't mind it for this example.
          [ 13309,      0,      5],
          [ 21527,      0,      6],
          [ 25699,      0,      5],
          [ 33965,      0,      6],
          [ 38088,      0,      5],
          [ 46357,      0,      6],
          [ 50478,      0,      5],
          [ 58785,      0,      6],
          [ 62929,      0,      5],
          [ 71182,      0,      6],
          [ 75315,      0,      5],
          [ 83511,      0,      6],
          [ 87646,      0,      5],
          [ 95866,      0,      6],
          [100035,      0,      5],
          [108247,      0,      6],
          [112425,      0,      5],
          [120693,      0,      6],
          [124814,      0,      5]], dtype=int64)
```

</details>

Now, from the way I understand the issue on the discourse forum; the user has something similar and wants to split each of the events in fix length epochs. If we consider my dataset as an example:
-> First regulation phase: 16 epochs with ID 5
-> First non-regulation phase: 8 epochs with ID 6
-> Second regulation phase: 16 more epochs with ID 5
-> Second non-regulation phase: 8 more epochs with ID 6
...

In my example, there is 10 alternation of regulation/non-regulation; thus leading to:
- 160 epochs with ID 5
- 80 epochs with ID 6
(if no rejection based on annotation is applied).

------------------------------------------------

**Problem:** This splitting has to know the duration of each event to split, in this example 8s and 16s.
Also, consider the case where we only want to do this splitting for one of the 2 events, e.g. 5, and ignore the second. This second use case may be useful in case you have events marking both phases (that last in time) and one-time event (that do not last in time). 

Maybe I completely misunderstood the discourse post :man_shrugging: but if not, what would be your take in implementing this?

hoechenberger, 2021-09-25:
I think there is a misunderstanding here. I `make_fixed_length_epochs` is supposed to cut continuous raw data into equidistant chunks, disregarding any events that might be present in the data.

mscheltienne, 2021-09-25:
Agree. But this is how I understood the post on discourse.
In my opinion, this is not something to add to `make_fix_length_epochs`, and sure we could add an argument to change the event ID from 1 to something else, but would that be of any help?

hoechenberger, 2021-09-25:
I'm not sure what the OP in the discourse posting did exactly, but they **do** say that they used `make_fixed_length_epochs` and the only issue they had was that the event ID was set to 1 🤔 

Your issue seems to be different… I think what you need is make use of the `start` and `stop` parameters of `make_fixed_length_events`, right? And then create a collection of same-length epochs (1 per time-locked event), which we later concatenate?

mscheltienne, 2021-09-27:
@hoechenberger Not sure either about the issue encountered by OP on discourse. I was ticking on:

> I have different events with different lengths of time. I want to make each second of each event as an epoch for classification purposes later.

For me, my description above is not really an issue. I can think of one or 2 different ways to do it, and your proposition would work like a charm. I was wondering if this scenario was encountered often, especially in online paradigms or in paradigms that have an alternation of long phases; in which case, a specific function to create this type of fixed-length epochs could be a good addition to MNE. As OP wrote on Discourse:

> Any suggestion that makes my life easier?

-----------------------------------------

Anyway, that said, and since I polluted a bit too much the discussion feed of this issue, I'll add the additional parameter to `make_fixed_length_epochs()` to change the event ID; at least this small upgrade won't be left out because of my post ;)

mscheltienne, 2021-09-29:
@hoechenberger To be closed. I don't know why it doesn't get closed despite the `fix` keyword in the PR.

hoechenberger, 2021-09-29:
You must mention "Fixes XXX" or "Closes XXX" in the PR message (first comment) for auto-closing

hoechenberger, 2021-09-29:
Closed via #9782

closed 2021-09-29 by hoechenberger

UmayKulsoom, 2023-02-08:
can I extract one single data from the fixed length epochs

hoechenberger, 2023-02-08:
Yes, like with any other ordinary epochs object. 

Please ask such usage questions on the forum at https://mne.discourse.group 

This issue tracker is intended mostly for bug reports and feature requests.

Thank you!