mne-tools/mne-python #9753: MAINT: Drop support for PySurfer + mayavi?

larsoner, 2021-09-16:
Now that:

1. Brain / PyVista / PyVistaQt has begun to mature;
2. PyVista backend has good notebook support;
3. `mayavi` has gone even more unmaintained than usual (I have had a PR open [for 3 months](https://github.com/enthought/mayavi/pull/1050) that fixes some bugs on latest `vtk` that hasn't been reviewed yet); and
4. PySurfer isn't too actively developed anymore

I propose that it's time that we drop support for it, as it lowers our maintenance burden: deprecate for 0.24, and in 0.25 remove the `mayavi` backend. We can leave the `set_3d_backend` infrastructure in place for future compatibility in case we want to migrate to a different 3D backend in the (probably distant) future.

I figured I'd ping @GuillaumeFavelier and @agramfort first since you've primarily been guiding the PyVista efforts. If you're on board, I'll ping the rest of the MNE steering committee since I think this is a potentially large enough change to warrant bothering everyone.

agramfort, 2021-09-16:
yes +1

but I'd like to kill this bird with the new coreg GUI so we're done for
good.

>

larsoner, 2021-09-16:
Oh right I forgot that the coreg GUI needed it.

I think there is actually another GUI -- `mne kit2fiff`. I have never used it, and I don't know how widely used it is...

agramfort, 2021-09-16:
no idea if mne kit2fiff is actually still used?

maybe @teonbrooks or @christianbrodbeck knows?

christianbrodbeck, 2021-09-17:
I am not aware of a replacement for the `kit2fiff` GUI, but since it only concerns a small subset of users maybe it could be kept as an special dependency, i.e. install mayavi if you want to use it?

teonbrooks, 2021-09-17:
for `kit2fiff` the conversion can be done programmatically without the GUI and we now have other tools such as `dig_montage`, its corresponding viz, and `coreg` to do inspection. It was a useful module to ensure that the markers were placed in the correct locations and that the corresponding files (headshape points, head coil points) were properly processed. I think making note of when it is last supported and point to the tutorial on using `dig_montage` would be what's needed.

agramfort, 2021-09-17:
I like this idea. Do you some basic script that would allow to replicate
the GUI feature?

>

hoechenberger, 2021-09-22:
+1000 on dropping Mayavi

teonbrooks, 2021-09-22:
> I like this idea. Do you some basic script that would allow to replicate the GUI feature?

i can try to tackle this on Friday

teonbrooks, 2021-10-08:
sorry, just had a time to come back to this. do we have access to the test data in an example script since this would require loading sqd/con files

larsoner, 2021-10-08:
https://github.com/mne-tools/mne-python/blob/main/mne/gui/tests/test_kit2fiff_gui.py#L17

larsoner, 2021-10-11:
@christianbrodbeck okay to remove these and go to scripting? Or are these still used by people / necessary for analyses?

christianbrodbeck, 2021-10-11:
@larsoner I am not aware of scripts that could fully replace the GUI, for example for quickly double checking and fixing the different marker files. But I'm just an occasional user now and could always create a separate `conda` environment with an older `mne` version. I don't know how many people still use it in KIT labs.

christianbrodbeck, 2021-10-11:
I can ask around in the labs I know if it will make a difference for this decision? Would you consider not deprecating it?

agramfort, 2021-10-12:
if it's used we need to propose a way out and not just kill the code. We
could move it to a new repo for example

>

larsoner, 2021-10-16:
> Is the biggest problem with Mayavi the fact that there is a largely absent maintainer? Are there other technical issues? ... What would help make things better?

I think that's the crux of the problem.

VTK as a big package breaks things from time to time, as happens in any large package (not trying to bad mouth them here). traits and traitsui make changes that break things from time to time, too. And because Mayavi wraps all VTK classes, it is good at exposing any new breaking issues with VTK -- which can be good for VTK and completeness testing (e.g., with finding a bug [that they fixed](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/6728)), but bad for downstream packages in that it can for example hang a Mayavi install, or manifest as segfaults, etc. At least this has been my observation working with `mayavi`.

So IMHO what `mayavi` needs is more regular/dedicated maintenance and CI testing against released and ideally prerelease/`dev` versions of VTK for example across all OSes (e.g., at least weekly). I suggested something like this [to the VTK people a year ago](https://gitlab.kitware.com/vtk/vtk/-/issues/17847) but nobody really jumped on the idea. Really I think it should be coupled with them doing automated deployment of prerelease/dev wheels, which *maybe* they could do without too much work since they [switched to automated wheel building](https://vtkpythonpackage.readthedocs.io/en/latest/Automated_wheels_building_with_scripts.html#automated-wheels-building-with-scripts).

So for us a lighter weight wrapper of VTK (e.g., `pyvista`) or making VTK calls directly, as we are doing more and more often, has broken stuff in our code less often than `mayavi`. We only have to fix the things that VTK breaks that we actually use, rather than fixing *anything* VTK has broken in a given release alongside anything traits/traitsui has broken compatibility-wise.

> Do you know anyone who would want to step in and help?

Not that I know of :( I personally try to make PRs when I have time, but I don't have the bandwidth to take on a core maintainer role at least.

prabhuramachandran, 2021-10-17:
@larsoner -- thanks for the response.  I agree with your analysis.  And yes, this is a real problem.  A major problem for me is that this has been the status for 20 years, bulk of the work has fallen on me and it has become harder and harder to find time or motivation.  That said, I still think that it is a useful project.  Another issue for me personally is that I have an almost "Debian stable" mentality and never update my Python environments for fear of breaking stuff so tend to live in my own cocoon -- that is the only way I can get my work done at all.  What I am going to do going forward is to put in a little time each week on Mayavi maintenance and the first goal is to work with some folks at Enthought to ensure that the CI is working and notifies us of issues.  I will take care of the core issues that I seem to be the only one comfortable with and others will help with setting up CI and other things that they are more comfortable managing.  Hopefully together this should be better.   Anyway, am hoping to do a better job. Thank you @larsoner for all your help these last few years and your patience.