mne-tools/mne-python #9759: Way to reset the raw's first_samp after cropping

DominiqueMakowski, 2021-09-18:
Using `raw.crop()` stores the first_samp value and can potentially create some issues down the line when externally manipulating the content (e.g., extracting channels as arrays, modifying them based on indices, and trying to re-add them, or find events in-them).

I would like to crop a Raw object and make it as if it was a true new Raw (i.e., discard any information relative to anything outside the new range).

I figured out that I can reset the `first_samp` argument like so: `raw._first_samps[0] = 0`.

It seems like `last_samp` is computed on-the-fly based on `first_samp`, so changing the latter mechanically propagates into the change.

However, but I wonder whether this is the only thing to change (and whether it might cause issues to simply reset it as is?).

Would it make sense to have a `reset = False` argument in `crop()` to do that?

Thanks a lot :)

agramfort, 2021-09-18:
>
> Using raw.crop() stores the first_samp value and can potentially create
> some issues down the line when externally manipulating the content (e.g.,
> extracting channels as arrays, modifying them based on indices, and trying
> to re-add them, or find events in-them).
>
I suggest you use then a RawArray for this. Don't hack private attributes.
The reason crop hacks like this is that you always want to keep track where
this chunk comes from in the original recording.

DominiqueMakowski, 2021-09-20:
Right, that makes sense, thanks!

closed 2021-09-20 by DominiqueMakowski