mne-tools/mne-python #9765: Epochs are not concatenated chronologically

jsosulski, 2021-09-21:
#### Describe the bug
Concatenate Epochs currently does not create chronologically ordered events.

#### Steps to reproduce

```python
import mne
import numpy as np

np.random.seed(10)
n_epochs, n_channels, n_timepoints = 2, 4, 21
ch_names = [f"ch_{i}" for i in range(n_channels)]
fake_data = 1e-6*np.random.random((n_epochs, n_channels, n_timepoints))
events = np.array([[1000, 0, 1], [20000, 0, 1]])
events2 = np.array([[1, 0, 1], [20, 0, 1]])

info = mne.create_info(ch_names=ch_names, sfreq=40)
epo = mne.EpochsArray(fake_data, info, events=events)
epo2 = mne.EpochsArray(np.copy(fake_data), info, events=events2)

epo_conc = mne.concatenate_epochs([epo, epo2])
assert all(np.diff(epo_conc.events[:, 0]) >= 0), "Not chronologically sorted events"
```

#### Expected results
Events are ordered.

#### Actual results
Events are not ordered.

#### Additional information
During concatenation, the offset is erroneously calculated on the current epoch to concatenate instead of the previous one. See the linked PR.

cbrnr, 2021-09-21:
I cannot reproduce. Here's what `epo_conc` looks like on 0.23.4:

```
>>> epo_conc.events
array([[ 1000,     0,     1],
       [20000,     0,     1],
       [20421,     0,     1],
       [20440,     0,     1]])
```

Is this on `main`?

cbrnr, 2021-09-21:
This was introduced in #9535.

jsosulski, 2021-09-21:
You are right. Yes sorry I am currently on main.

PR #9766 (2021-09-23): FIX: chronological order of concatenated epochs
#### Reference issue
Fixes #9765 

#### What does this implement/fix?
Fix a bug in concatenate which could create epochs that are not ordered chronologically.
closed 2021-09-23 by larsoner