mne-tools/mne-python #9769: Error using mne.read_evoked_fieldtrip: Key Error Data Name

emyskog, 2021-09-22:
I have a .mat file with averages for different conditions that were computed in Fieldtrip. I am trying to use mne.read_evoked_fieldtrip, but when I run this code I get an `error: KeyError:  's2ICAOK_OFF_D_T1'`. I read the epochs with no issue.

`import matplotlib.pyplot as plt

from sklearn.pipeline import make_pipeline
from sklearn.preprocessing import StandardScaler
from sklearn.linear_model import LogisticRegression

import mne
from mne.datasets import sample
from mne.decoding import GeneralizingEstimator

print(__doc__)

data_epochs = mne.read_epochs_fieldtrip('s21ICAOK.mat', None,data_name='s21ICAOK')
data_on = mne.read_evoked_fieldtrip('s2ICAOK_OFF_D_T1.mat', None, data_name='s2ICAOK_OFF_D_T1')`

agramfort, 2021-09-24:
are you sure data_name is correct? how did you save the file with fieldtrip? did you manually edit the matlab structure?

transpersonify, 2022-11-02:
Hello,
When I load data using mne.read_evoked_fieldtrip, I am not able to read times, I added tmin value in the function by reading it from the fieldtrip structure as follows. 

`    time= np.squeeze(ft_struct['time'])
    tmin = time[0]
    
    info = _create_info(ft_struct, info)  # create info structure
    data_evoked = ft_struct['avg']  # create evoked data
    evoked = EvokedArray(data_evoked, info, comment=comment, tmin=tmin)

`
This works fine and I am able to plot the data as expected. However, to do baseline correction in this evoked data, I had to add one hack to read_evoked_fieldtrip function:
```
    if(comment==None):
        comment = ''
```
If I don't do this any evoked method that reads the comment field gives error: object of type 'NoneType' has no len().
This error is coming from the following patch:    

```
 def __repr__(self):  # noqa: D105
        max_comment_length = 1000
        if len(self.comment) > max_comment_length:
            comment = self.comment[:max_comment_length]
            comment += "..."
```
This is just for information.

transpersonify, 2022-11-02:
Hello,
When I load data using mne.read_evoked_fieldtrip, I am not able to read times, I added tmin value in the function by reading it from the fieldtrip structure as follows.

``` time= np.squeeze(ft_struct['time'])
tmin = time[0]

info = _create_info(ft_struct, info)  # create info structure
data_evoked = ft_struct['avg']  # create evoked data
evoked = EvokedArray(data_evoked, info, comment=comment, tmin=tmin)
```
This works fine and I am able to plot the data as expected. However, to do baseline correction in this evoked data, I had to add one hack to read_evoked_fieldtrip function:

    if(comment==None):
        comment = ''

If I don't do this any evoked method that reads the comment field gives error: object of type 'NoneType' has no len().
This error is coming from the following patch:

 def __repr__(self):  # noqa: D105
        max_comment_length = 1000
        if len(self.comment) > max_comment_length:
            comment = self.comment[:max_comment_length]
            comment += "..."

This is just for information.

drammock, 2022-11-02:
@transpersonify I've migrated your comment into a new issue, #11296. Please post any follow-ups there

drammock, 2022-11-02:
closing, as >1 year with no response from OP.

closed 2022-11-02 by drammock