mne-tools/mne-python #9774: BUG: Pooch fails to download on CircleCI

larsoner, 2021-09-22:
Looks like we have a problem, presumably from #9742:

https://app.circleci.com/pipelines/github/mne-tools/mne-python/10239/workflows/959aeb3e-ddbd-45a8-9b11-d8915f6ae1f6/jobs/34623
```
Downloading file 'hf_sef_evoked.tar.gz' from 'https://zenodo.org/record/3523071/files/hf_sef_evoked.tar.gz' to '/home/circleci/mne_data'.
100%|███████████████████████████████████████| 766M/766M [00:00<00:00, 3.35TB/s]
Extracting 'hf_sef/MEG' from '/home/circleci/mne_data/hf_sef_evoked.tar.gz' to '/home/circleci/mne_data'
Traceback (most recent call last):
  File "/home/circleci/.local/lib/python3.8/site-packages/pooch/processors.py", line 219, in _extract_file
    output.write(data_file.read())
AttributeError: 'NoneType' object has no attribute 'read'
```
@adam2392 I can replicate locally with:
```
$ python -c "import mne; mne.datasets.hf_sef.data_path(force_update=True, verbose=True)"
Using default location ~/mne_data for HF_SEF...
Downloading file 'hf_sef_evoked.tar.gz' from 'https://zenodo.org/record/3523071/files/hf_sef_evoked.tar.gz' to '/Users/larsoner/mne_data'.
100%|███████████████████████████████████████| 766M/766M [00:00<00:00, 1.79TB/s]
Extracting 'hf_sef/MEG' from '/Users/larsoner/mne_data/hf_sef_evoked.tar.gz' to '/Users/larsoner/mne_data'
Traceback (most recent call last):
  File "/Users/larsoner/opt/miniconda3/lib/python3.8/site-packages/pooch/processors.py", line 219, in _extract_file
    output.write(data_file.read())
AttributeError: 'NoneType' object has no attribute 'read'
```
Can you look into it?

adam2392, 2021-09-22:
Confirmed able to reproduce error locally, but not sure why the CI works on: https://github.com/mne-tools/mne-python/pull/9763 then?

Looking into why pooch is even raising the error because it seemed to work w/ our unit tests...

larsoner, 2021-09-22:
> Confirmed able to reproduce error locally, but not sure why the CI works on: #9763 then?

Datasets are only downloaded if needed. CircleCI caches invalidate at different times for different datasets. So my guess is that it was always broken since #9742 went in, but us noticing it was just delayed until an actual download had to happen (in this case, because some CircleCI cache expired, and then it went to download, and it failed).

adam2392, 2021-09-22:
So I'm able to fix it by changing hf_sef pooch processor to just 

Untar(path)

From originally:

Untar(path, members=...)

Idk if we need fine grained extraction of the dataset based on the MEG, subjects and SSS directory right? We want all 3 subdirectories so this should be a safe change?

Seems weird and might be a bug on pooch end imo so I can file an upstream issue later.

larsoner, 2021-09-22:
To check if that fix is correct at our end, can you try downloading the dataset before #9742, examine the contents, delete the directory, download it with #9742 plus your proposed fix, and make sure the contents are the same (i.e., same set of files)? We don't want to include a bunch of extra junk that's in there (if it's in there), I wonder if that's why we wanted a `members=...`

adam2392, 2021-09-23:
[image]
[image]

Looks like the contents are the same w/ the latest push to #9763

larsoner, 2021-09-23:
To me it looks like a bug that the old one is `HF_SEF` and the new one is `hf_sef`. On `macOS` and Windows this will not matter, but on Linux it will.

Also it does not seem to work for me:
```
$ python -c "import mne; print(mne.datasets.hf_sef.data_path(verbose=True))"
/home/larsoner/mne_data/HF_SEF
$ git checkout -b fetchapi adam2392/fetchapi
$ python -c "import mne; print(mne.datasets.hf_sef.data_path(verbose=True))"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<decorator-gen-510>", line 22, in data_path
  File "/home/larsoner/python/mne-python/mne/datasets/hf_sef/hf_sef.py", line 78, in data_path
    return fetch_dataset(dataset_params=dataset_params, processor=processor,
  File "/home/larsoner/python/mne-python/mne/datasets/fetch.py", line 130, in fetch_dataset
    names = [params['dataset_name'] for params in dataset_params]
  File "/home/larsoner/python/mne-python/mne/datasets/fetch.py", line 130, in <listcomp>
    names = [params['dataset_name'] for params in dataset_params]
KeyError: 'dataset_name'
```

adam2392, 2021-09-23:
Hmmm working on a fix here.

So the issue with `HF_SEF` vs `hf_sef` seems to be that the folder in the tar file is `hf_sef`, but our desired folder_name is actually `HF_SEF`. That is probably the reason for these LOC: https://github.com/mne-tools/mne-python/pull/9742#discussion_r714333528

However, those LOC are what break our circleCI. https://github.com/fatiando/pooch/blob/7a236035009cb3b30288f023206aece64ea397e7/pooch/processors.py#L204-L221 assume that the underlying `member` is a file I think, rather than a folder. Seems like a bug for pooch. 

I think to match the casing of the folder name and get the untar in the right place, we can do a hack on our end:
1. untar the file naively
2. rename the folder name to HF_SEF

adam2392, 2021-09-23:
Upstream issue: https://github.com/fatiando/pooch/issues/264

larsoner, 2021-09-23:
Agreed I would just rename. Although really maybe we can live with `hf_sef` being the new name rather than fight with it. I don't think that dataset is downloaded super often...

Maybe just fix the `names = [params['dataset_name'] for params in dataset_params]` bug and then we'll merge #9763?

adam2392, 2021-09-23:
Sounds good. I fixed it now and also renamed the `HF_SEF` folder to `hf_sef`.

PR #9763 (2021-09-25): [MRG, MAINT] Fetching Datasets Public API "fetch_dataset" used by all internal MNE datasets now
#### Reference issue
Closes #9736 
Closes #9774

#### What does this implement/fix?
This adds the public function `mne.datasets.fetch_dataset` and refactors the existing MNE datasets to use that function.

This also refactors `_data_path` function to be more generic.
closed 2021-09-25 by larsoner