mne-tools/mne-python #9783: Can Epochs.plot have a "show_scale_bar" parameter similarly to Raw.plot?

adam2392, 2021-09-27:
#### Describe the new feature or enhancement
I'm plotting some Epoch data and I noticed that the parameters are slightly different from Raw.plot. Specifically, the scale bar is not able to be shown in Epoch.plot. However, I'm not sure why not?

#### Describe your proposed implementation
Perhaps add some of the parameters into the Epochs.plot that is available in Raw.plot

#### Additional comments
https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.plot

https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.plot

drammock, 2021-09-27:
I think it's not in the method signature but you can still enable it once the plot is open by pressing `s`?

adam2392, 2021-09-27:
I tend to only use matplotlib to plot inline so have to enable it programmatically. 

The interactive plotting always crashes on my Mac :p.

cbrnr, 2021-09-28:
I think we should make all options available as parameters, including the scale bar, but there are many more toggles that are not exposed.

hoechenberger, 2021-09-28:
+10 on adding this

closed 2021-10-08 by agramfort