mne-tools/mne-python #9788: Getting RuntimeError:  Render window interactor must be initialized before processing events.

balandongiv, 2021-09-28:
#### Describe the bug
When running the tutorial [Compute coherence in source space using a MNE inverse solution](https://mne.tools/mne-connectivity/stable/auto_examples/mne_inverse_coherence_epochs.html), the compiler return an error 

> RuntimeError: Render window interactor must be initialized before processing events.

at line `brain.show_view('lateral')`

#### Actual results
The full tracebackl

```
---------------------------------------------------------------------------

RuntimeError                              Traceback (most recent call last)

/tmp/ipykernel_23683/1897321839.py in <module>
    139 print('t1')
    140 # Now we can visualize the coherence using the plot method.
--> 141 brain = coh_stc.plot('sample', 'inflated', 'both',
    142                      time_label='Coherence %0.1f Hz',
    143                      subjects_dir=subjects_dir,

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/source_estimate.py in plot(self, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, subjects_dir, figure, views, colorbar, clim, cortex, size, background, foreground, initial_time, time_unit, backend, spacing, title, show_traces, src, volume_options, view_layout, add_data_kwargs, brain_kwargs, verbose)
    649              src=None, volume_options=1., view_layout='vertical',
    650              add_data_kwargs=None, brain_kwargs=None, verbose=None):
--> 651         brain = plot_source_estimates(
    652             self, subject, surface=surface, hemi=hemi, colormap=colormap,
    653             time_label=time_label, smoothing_steps=smoothing_steps,

<decorator-gen-153> in plot_source_estimates(stc, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, subjects_dir, figure, views, colorbar, clim, cortex, size, background, foreground, initial_time, time_unit, backend, spacing, title, show_traces, src, volume_options, view_layout, add_data_kwargs, brain_kwargs, verbose)

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/viz/_3d.py in plot_source_estimates(stc, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, subjects_dir, figure, views, colorbar, clim, cortex, size, background, foreground, initial_time, time_unit, backend, spacing, title, show_traces, src, volume_options, view_layout, add_data_kwargs, brain_kwargs, verbose)
   1779     else:
   1780         with use_3d_backend(backend):
-> 1781             return _plot_stc(
   1782                 stc, overlay_alpha=alpha, brain_alpha=alpha,
   1783                 vector_alpha=alpha, cortex=cortex, foreground=foreground,

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/viz/_3d.py in _plot_stc(stc, subject, surface, hemi, colormap, time_label, smoothing_steps, subjects_dir, views, clim, figure, initial_time, time_unit, background, time_viewer, colorbar, transparent, brain_alpha, overlay_alpha, vector_alpha, cortex, foreground, size, scale_factor, show_traces, src, volume_options, view_layout, add_data_kwargs, brain_kwargs)
   1861                              'mayavi backend')
   1862     with warnings.catch_warnings(record=True):  # traits warnings
-> 1863         brain = Brain(**kwargs)
   1864     del kwargs
   1865 

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/viz/_brain/_brain.py in __init__(self, subject_id, hemi, surf, title, cortex, alpha, size, background, foreground, figure, subjects_dir, views, offset, show_toolbar, offscreen, interaction, units, view_layout, silhouette, theme, show)
    526                         decimate=self._silhouette["decimate"],
    527                     )
--> 528                 self._renderer.set_camera(**views_dicts[h][v])
    529 
    530         self.interaction = interaction

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/viz/backends/_pyvista.py in set_camera(self, azimuth, elevation, distance, focalpoint, roll, reset_camera, rigid)
    597                    focalpoint='auto', roll=None, reset_camera=True,
    598                    rigid=None):
--> 599         _set_3d_view(self.figure, azimuth=azimuth, elevation=elevation,
    600                      distance=distance, focalpoint=focalpoint, roll=roll,
    601                      reset_camera=reset_camera, rigid=rigid)

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/mne/viz/backends/_pyvista.py in _set_3d_view(figure, azimuth, elevation, focalpoint, distance, roll, reset_camera, rigid)
    994         figure.plotter.camera.SetRoll(figure.plotter.camera.GetRoll() + roll)
    995 
--> 996     figure.plotter.update()
    997     _process_events(figure.plotter)
    998 

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/pyvista/plotting/plotting.py in update(self, stime, force_redraw)
   1401             update_rate = self.iren.get_desired_update_rate()
   1402             if (curr_time - Plotter.last_update_time) > (1.0/update_rate):
-> 1403                 self.right_timer_id = self.iren.create_repeating_timer(stime)
   1404                 self.render()
   1405                 Plotter.last_update_time = curr_time

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/pyvista/plotting/render_window_interactor.py in create_repeating_timer(self, stime)
    607         timer_id = self.interactor.CreateRepeatingTimer(stime)
    608         if hasattr(self.interactor, 'ProcessEvents'):
--> 609             self.process_events()
    610         else:
    611             self.interactor.Start()

~/anaconda3/envs/source_space_mne/lib/python3.9/site-packages/pyvista/plotting/render_window_interactor.py in process_events(self)
    629         # Note: This is only available in VTK 9+
    630         if not self.initialized:
--> 631             raise RuntimeError('Render window interactor must be initialized '
    632                                'before processing events.')
    633         self.interactor.ProcessEvents()

RuntimeError: Render window interactor must be initialized before processing events.
```

#### Additional information:

Ubuntu 18.04
Issue occur for both Python 3.7 and 3.9.

Package             Version
------------------- -----------
aiohttp             3.7.4.post0
appdirs             1.4.4
argcomplete         1.12.3
argon2-cffi         20.1.0
async-generator     1.10
async-timeout       3.0.1
attrs               21.2.0
autobahn            21.3.1
Automat             20.2.0
backcall            0.2.0
bleach              4.0.0
certifi             2021.5.30
cffi                1.14.6
chardet             4.0.0
constantly          15.1.0
cryptography        3.4.8
cycler              0.10.0
debugpy             1.4.1
decorator           5.0.9
defusedxml          0.7.1
entrypoints         0.3
hyperlink           21.0.0
idna                3.2
imageio             2.9.0
importlib-metadata  4.8.1
incremental         21.3.0
ipykernel           6.2.0
ipython             7.27.0
ipython-genutils    0.2.0
ipywidgets          7.6.4
jedi                0.18.0
Jinja2              3.0.1
joblib              1.0.1
jsonschema          3.2.0
jupyter             1.0.0
jupyter-client      7.0.1
jupyter-console     6.4.0
jupyter-core        4.7.1
jupyterlab-pygments 0.1.2
jupyterlab-widgets  1.0.0
kiwisolver          1.3.2
MarkupSafe          2.0.1
matplotlib          3.4.3
matplotlib-inline   0.1.2
meshio              4.4.6
mistune             0.8.4
mne                 0.23.4
multidict           5.1.0
nbclient            0.5.3
nbconvert           6.1.0
nbformat            5.1.3
nest-asyncio        1.5.1
notebook            6.4.3
numpy               1.21.2
packaging           21.0
pandocfilters       1.4.3
parso               0.8.2
pexpect             4.8.0
pickleshare         0.7.5
Pillow              8.3.2
pip                 21.0.1
prometheus-client   0.11.0
prompt-toolkit      3.0.17
ptyprocess          0.7.0
pycparser           2.20
Pygments            2.10.0
pyparsing           2.4.7
PyQt5               5.15.4
PyQt5-Qt5           5.15.2
PyQt5-sip           12.9.0
pyrsistent          0.17.3
python-dateutil     2.8.2
pyvista             0.32.1
pyzmq               22.2.1
qtconsole           5.1.1
QtPy                1.10.0
scikit-learn        1.0
scipy               1.7.1
scooby              0.5.7
Send2Trash          1.8.0
setuptools          58.0.4
six                 1.16.0
sklearn             0.0
terminado           0.9.4
testpath            0.5.0
threadpoolctl       2.2.0
tornado             6.1
tqdm                4.62.3
traitlets           5.1.0
Twisted             21.7.0
txaio               21.2.1
typing-extensions   3.10.0.2
vtk                 9.0.3
wcwidth             0.2.5
webencodings        0.5.1
wheel               0.37.0
widgetsnbextension  3.5.1
wslink              1.0.7
yarl                1.6.3
zipp                3.5.0
zope.interface      5.4.0

agramfort, 2021-09-28:
can you provide a full code snippet to replicate the crash?

>

balandongiv, 2021-09-28:
Hi @agramfort , it is exactly as provided in the website. 

https://mne.tools/mne-connectivity/stable/auto_examples/mne_inverse_coherence_epochs.html
```

"""
==============================================================
Compute coherence in source space using a MNE inverse solution
==============================================================

This example computes the coherence between a seed in the left
auditory cortex and the rest of the brain based on single-trial
MNE-dSPM inverse solutions.
"""
# Author: Martin Luessi <mluessi@nmr.mgh.harvard.edu>
#
# License: BSD (3-clause)

import numpy as np

import mne
from mne.datasets import sample
from mne.minimum_norm import (apply_inverse, apply_inverse_epochs,
                              read_inverse_operator)
from mne.connectivity import seed_target_indices, spectral_connectivity

print(__doc__)

###############################################################################
# Read the data
# -------------
#
# First we'll read in the sample MEG data that we'll use for computing
# coherence between channels. We'll convert this into epochs in order to
# compute the event-related coherence.

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
fname_event = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
label_name_lh = 'Aud-lh'
fname_label_lh = data_path + '/MEG/sample/labels/%s.label' % label_name_lh

event_id, tmin, tmax = 1, -0.2, 0.5
method = "dSPM"  # use dSPM method (could also be MNE or sLORETA)

# Load data.
inverse_operator = read_inverse_operator(fname_inv)
label_lh = mne.read_label(fname_label_lh)
raw = mne.io.read_raw_fif(fname_raw)
events = mne.read_events(fname_event)

# Add a bad channel.
raw.info['bads'] += ['MEG 2443']

# pick MEG channels.
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False, eog=True,
                       exclude='bads')

# Read epochs.
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0),
                    reject=dict(mag=4e-12, grad=4000e-13, eog=150e-6))

###############################################################################
# Choose channels for coherence estimation
# ----------------------------------------
#
# Next we'll calculate our channel sources. Then we'll find the most active
# vertex in the left auditory cortex, which we will later use as seed for the
# connectivity computation.

snr = 3.0
lambda2 = 1.0 / snr ** 2
evoked = epochs.average()
stc = apply_inverse(evoked, inverse_operator, lambda2, method,
                    pick_ori="normal")

# Restrict the source estimate to the label in the left auditory cortex.
stc_label = stc.in_label(label_lh)

# Find number and index of vertex with most power.
src_pow = np.sum(stc_label.data ** 2, axis=1)
seed_vertno = stc_label.vertices[0][np.argmax(src_pow)]
seed_idx = np.searchsorted(stc.vertices[0], seed_vertno)  # index in orig stc

# Generate index parameter for seed-based connectivity analysis.
n_sources = stc.data.shape[0]
indices = seed_target_indices([seed_idx], np.arange(n_sources))

###############################################################################
# Compute the inverse solution for each epoch. By using "return_generator=True"
# stcs will be a generator object instead of a list. This allows us so to
# compute the coherence without having to keep all source estimates in memory.

snr = 1.0  # use lower SNR for single epochs
lambda2 = 1.0 / snr ** 2
stcs = apply_inverse_epochs(epochs, inverse_operator, lambda2, method,
                            pick_ori="normal", return_generator=True)

###############################################################################
# Compute the coherence between sources
# -------------------------------------
#
# Now we are ready to compute the coherence in the alpha and beta band.
# fmin and fmax specify the lower and upper freq. for each band, respectively.
#
# To speed things up, we use 2 parallel jobs and use mode='fourier', which
# uses a FFT with a Hanning window to compute the spectra (instead of
# a multitaper estimation, which has a lower variance but is slower).
# By using faverage=True, we directly average the coherence in the alpha and
# beta band, i.e., we will only get 2 frequency bins.

fmin = (8., 13.)
fmax = (13., 30.)
sfreq = raw.info['sfreq']  # the sampling frequency

coh, freqs, times, n_epochs, n_tapers = spectral_connectivity(
    stcs, method='coh', mode='fourier', indices=indices,
    sfreq=sfreq, fmin=fmin, fmax=fmax, faverage=True, n_jobs=1)

print('Frequencies in Hz over which coherence was averaged for alpha: ')
print(freqs[0])
print('Frequencies in Hz over which coherence was averaged for beta: ')
print(freqs[1])

###############################################################################
# Generate coherence sources and plot
# -----------------------------------
#
# Finally, we'll generate a SourceEstimate with the coherence. This is simple
# since we used a single seed. For more than one seed we would have to choose
# one of the slices within ``coh``.
#
# .. note:: We use a hack to save the frequency axis as time.
#
# Finally, we'll plot this source estimate on the brain.

tmin = np.mean(freqs[0])
tstep = np.mean(freqs[1]) - tmin
coh_stc = mne.SourceEstimate(coh, vertices=stc.vertices, tmin=1e-3 * tmin,
                             tstep=1e-3 * tstep, subject='sample')

# Now we can visualize the coherence using the plot method.
brain = coh_stc.plot('sample', 'inflated', 'both',
                     time_label='Coherence %0.1f Hz',
                     subjects_dir=subjects_dir,
                     clim=dict(kind='value', lims=(0.25, 0.4, 0.65)))
brain.show_view('lateral')

```

But, I suspect the latest vtk 9.0.3 is the culprit.

larsoner, 2021-09-28:
It could be VTK 9.0.3. On your 3.7 Python you can use vtk==9.0.1 and see if it helps.

I don't see pyvistaqt installed, so I'm assuming you want this to run in the notebook? In theory it should auto-detect this, but try putting `mne.viz.set_3d_backend('notebook')` at the top of the example to see if it helps

balandongiv, 2021-09-29:
Thanks for the time entertain this issue @larsoner.

There are two question from your comment above

1) May I know at which top position do you mean to place the `mne.viz.set_3d_backend('notebook')`. Place the line just below `import mne` still give me the error

> RuntimeError: Render window interactor must be initialized before processing events.

Currently, I place the   `mne.viz.set_3d_backend('notebook')` as shown below.

```
import numpy as np

import mne
mne.viz.set_3d_backend('notebook')
from mne.datasets import sample
from mne.minimum_norm import (apply_inverse, apply_inverse_epochs,
                              read_inverse_operator)
from mne.connectivity import seed_target_indices, spectral_connectivity

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
fname_event = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
label_name_lh = 'Aud-lh'
fname_label_lh = data_path + '/MEG/sample/labels/%s.label' % label_name_lh
```

 2) May I know there exist any setting to halt program execution until the vtk  is closed when using `.py` to render the plot.

brain.show_view('lateral',block=True)?

This is because, the 3D viz `ui` open and close in instance when using `.py` to render the plot.

larsoner, 2021-09-29:
> May I know at which top position do you mean to place the mne.viz.set_3d_backend('notebook'). Placing this line just below import mne still produce the error

Yeah that's what I had in mind. This will only work if you're running in a jupyter notebook, though. If you're in plain Python or IPython it shouldn't work. (In principle it should raise an error, but it might not...)

> May I know whether there exist any setting to halt program execution until the vtk is closed when using .py to render the plot.

I usually run Python in interactive mode, e.g., `python -i script.py` for this. Or open `ipython` and do `%run script.py`, etc.

balandongiv, 2021-10-01:
> This will only work if you're running in a jupyter notebook,

The issue still remain,this maybe as I'm using Pycharm in-build Notebook

>  usually run Python in interactive mode, e.g., python -i script.py for this. Or open ipython and do %run script.py, etc.

Thanks for the trick.

Please advice whether this issue should be closed or otherwise

larsoner, 2021-10-01:
I think we're good for now, thanks for letting us know it's working @balandongiv !

closed 2021-10-01 by larsoner