mne-tools/mne-python #9790: Failed to import dat files.

sam1o1, 2021-09-28:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should ask for help on the
[MNE Forum](https://mne.discourse.group).

Our GitHub issue tracker is only used to report bugs and suggest improvements
to MNE-Python. For any other questions, please use the forum.
Usage questions that are posted as GitHub issues are usually closed without
being answered. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
please **delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block (see
[the GitHub guide to markdown](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown)
for help with issue formatting). Alternatively, you can put your MWE in a
[public gist](https://gist.github.com) and link to it in this issue.

#### Describe the bug
I was working on EEG data that was stored as dat files from the neurosky sensor. When I tried to import the data using `mne.channels.read_dig_dat` I got an error. 

#### Steps to reproduce
*Replace this text with a code snippet or minimal working example [MWE] to
replicate your problem, using one of the [built-in datasets], preferably the
one called [sample]. If you can't replicate on a built-in dataset, provide also
a link to a small, anonymized portion of your data that does yield the error.*

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
*Replace this text with a description of what you expected to happen.*

#### Actual results
*Replace this text with the actual output, traceback, screenshot, or other
description of the results.*

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*

sappelhoff, 2021-09-28:
`mne.channels.read_dig_dat` is a function to read electrode locations from some "Compumedics" file format, see:

https://github.com/mne-tools/mne-python/blob/2f6d6b4667af1a3214e1a84ae21f8109429770a3/mne/channels/montage.py#L463-L471

If you want to read EEG data, please have a look at this documentation page: https://mne.tools/dev/auto_tutorials/io/20_reading_eeg_data.html it lists all currently supported file formats. If the format you want to read is not yet supported, it'd be helpful if you could supply as much information (including links) as possible, so that people can get an impression of how easy/difficult/useful it'd be to implement reading support for your data format.

closed 2021-09-28 by agramfort

sam1o1, 2021-09-28:
I tried to use mne.io.read_raw_persyst  and It failed too to load .dat files.

sappelhoff, 2021-09-29:
I recommend you do some research into what data format your system (you mentioned "neurosky sensor") produces, and then help us with lots of info on that. :-)