mne-tools/mne-python #9791: TypeError raised in the IPython console when representing a RawArray.

mscheltienne, 2021-09-29:
I had this minor bug a couple of times already, and aside from being annoying, it's harmless.
When representing a created `mne.io.RawArray` in the IPython console, it raises a TypeError... before giving the correct representation. 

```
import mne
import numpy as np

info = mne.create_info(['STI'], sfreq=512, ch_types='stim')
stim_ch = mne.io.RawArray(np.zeros(shape=(1, 1000)), info)
```

Call representation:

```
In[2]: stim_ch
Traceback (most recent call last):

  File "/Users/scheltie/Documents/pyvenv/neurotin/lib/python3.8/site-packages/IPython/core/formatters.py", line 345, in __call__
    return method()

  File "/Users/scheltie/Documents/pyvenv/neurotin/lib/python3.8/site-packages/mne/io/base.py", line 1763, in _repr_html_
    basenames = [os.path.basename(f) for f in self._filenames]

  File "/Users/scheltie/Documents/pyvenv/neurotin/lib/python3.8/site-packages/mne/io/base.py", line 1763, in <listcomp>
    basenames = [os.path.basename(f) for f in self._filenames]

  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/posixpath.py", line 142, in basename
    p = os.fspath(p)

TypeError: expected str, bytes or os.PathLike object, not NoneType

Out[2]: <RawArray | 1 x 1000 (2.0 s), ~14 kB, data loaded>
```

A normal `Raw` object loaded from disk does not raise any error:

```
raw
Out[4]: <Raw | 1-online-neurofeedback-eegoSports 000479-raw.fif, 67 x 135497 (264.6 s), ~69.4 MB, data loaded>
```

Just guessing that since `stim_ch.filenames` is equal to `(None,)`, something breaks in the representation method of `BaseRaw`.

mscheltienne, 2021-09-29:
Never mind... should have run that on the dev version. It works, so probably already fixed. Probably in #9404.

closed 2021-09-29 by mscheltienne

larsoner, 2021-09-29:
Argh I thought we backported that but we didn't. If we end up wanting another point release before cutting 0.24 in early Nov then we should include #9404. Thanks for investigating @mscheltienne

mscheltienne, 2021-09-29:
@larsoner It might have been backported, I didn't check and the environment I was running was probably with 0.23.0.

larsoner, 2021-09-29:
I checked, it was not https://github.com/mne-tools/mne-python/blob/maint/0.23/mne/io/base.py#L1763