mne-tools/mne-python #9792: ENH: Add Neuromag phantom subject

larsoner, 2021-09-29:
I created an incomplete FreeSurfer subject directory for the Neuromag phantom that contains the head surface and CT-as-MRI T1.mgz. With this, you can do a plot for [this](https://mne.tools/dev/auto_tutorials/inverse/80_brainstorm_phantom_elekta.html?highlight=phantom) example that would look like:
```
import mne
import numpy as np

data_path = mne.datasets.brainstorm.bst_phantom_elekta.data_path()
info = mne.io.read_info(data_path + '/kojak_all_200nAm_pp_no_chpi_no_ms_raw.fif')
sphere = mne.make_sphere_model((0., 0., 0.), 0.08)
trans = mne.transforms.Transform('head', 'mri', np.eye(4))
fig = mne.viz.plot_alignment(
    info, trans, 'phantom', bem=sphere, surfaces=('inner_skull', 'head-dense'),
    coord_frame='meg', show_axes=True, mri_fiducials=True, subjects_dir='.',
    dig=True)
fig.plotter.screenshot('alignment.png')
```

[image]

FYI for folks who haven't seen it, this is what [the phantom](https://neuroimage.usc.edu/brainstorm/Tutorials/PhantomElekta#The_phantom) actually looks like:

<details>

[image]

</details>

Options:

1. Add to `bst_phantom_elekta`. I don't like this because the phantom is more generally usable than this, and it's not part of the data Brainstorm actually distributes, either.
2. Add to `mne-misc-data`. It's small so this is doable, but not the best organization-wise.
3. Add a `mne.datasets.fetch_phantom`. It's more stuff in that namespace (not great), but probably the cleanest because you can tell it explicitly which `subjects_dir` to put the subject into. It's also easier to update/add files in the future and for users to get them, similar to how `fetch_fsaverage` works.
4. Don't distribute this, nobody actually needs it (though I do, other MEG folks might, and it would make the bst example nicer!)

I prefer option (3), any thoughts?

agramfort, 2021-09-29:
+1 for (3). I think it's a great addition !

>

drammock, 2021-09-29:
> +1 for (3). I think it's a great addition !

agreed

rob-luke, 2021-09-30:
But maybe `fetch_meg_phantom`? we have phantoms for fNIRS and EEG too.

larsoner, 2021-09-30:
Or maybe `fetch_phantom(kind='neuromag')`

larsoner, 2021-09-30:
(this could mirror the design of get_phantom_dipoles, which would be nice)

PR #9796 (2021-10-01): MRG, ENH: Add Otaniemi phantom based on CT
Turns out the CT I already had was for the Otaniemi phantom. I'm waiting to hear back if we can get a CT of the newer phantom for the VectorView / TRIUX systems.

- [x] Add Otaniemi phantom
- [x] Add tests
- [ ] ~~Add VectorView phantom~~
- [x] Update VectorView example
- [x] Update latest.inc

Closes #9792
closed 2021-10-01 by agramfort