mne-tools/mne-python #9797: Problem using mne.decoding.receptivefield

moonj94, 2021-09-30:
#### Describe the bug
When I try to run receptivefield.fit(X,Y), I get the following error in my terminal and spyder restarts the kernel
`ERROR:traitlets:No such comm target registered: jupyter.widget`

I have also tried:

```
jupyter nbextension install --py widgetsnbextension --user

jupyter nbextension enable --py widgetsnbextension
```
#### Steps to reproduce

This is the code that would reproduce the problem. Versions:
```
Spyder                   5.1.5
MNE                        23.4
jupyter                   1.0.0           
jupyter_client            6.1.12             
jupyter_console           6.4.0              
jupyter_core              4.7.1            
jupyterlab_pygments       0.1.2                      
jupyterlab_widgets        1.0.0 
```
```
tmin,tmax=0,Y.shape[0]/256
n_delays=int((tmax-tmin)*sfreq)+2
n_splits=3
cv=KFold(n_splits)
n_channels=64
speech=np.random.randn(256,1)
Y=np.random.randn(256,64)
rf=ReceptiveField(tmin,tmax,sfreq,feature_names=['envelope'],estimator=1,scoring='corrcoef')
coefs = np.zeros((n_splits, n_channels, n_delays))
scores = np.zeros((n_splits, n_channels))
for ii, (train, test) in enumerate(cv.split(speech)):
    print('split %s / %s' % (ii + 1, n_splits))
    rf.fit(speech[train], Y[train])
    scores[ii] = rf.score(speech[test], Y[test])
    # coef_ is shape (n_outputs, n_features, n_delays). we only have 1 feature
    coefs[ii] = rf.coef_[:, 0, :]
times = rf.delays_ / float(rf.sfreq)

# Average scores and coefficients across CV splits
mean_coefs = coefs.mean(axis=0)
mean_scores = scores.mean(axis=0)

# Plot mean prediction scores across all channels
fig, ax = plt.subplots()
ix_chs = np.arange(n_channels)
ax.plot(ix_chs, mean_scores)
ax.axhline(0, ls='--', color='r')
ax.set(title="Mean prediction score", xlabel="Channel", ylabel="Score ($r$)")
mne.viz.tight_layout()
```

#### Additional information
output of mne.sys_info():

```
Platform:      macOS-10.16-x86_64-i386-64bit
Python:        3.7.10 | packaged by conda-forge | (default, Sep 13 2021, 19:35:18)  [Clang 11.1.0 ]
Executable:    /Users/jaemoon/opt/anaconda3/envs/phdf/bin/python
CPU:           i386: 12 cores
Memory:        16.0 GB

mne:           0.23.4
numpy:         1.19.5 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.7.1
matplotlib:    3.4.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:       0.24.2
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.8.1
dipy:          1.3.0
cupy:          Not found
pandas:        1.3.3
mayavi:        Not found
pyvista:       0.32.1 {pyvistaqt=0.5.0, OpenGL 4.1 ATI-4.6.20 via AMD Radeon Pro 560X OpenGL Engine}
vtk:           9.0.1
PyQt5:         5.12.3
```

agramfort, 2021-10-01:
the MNE code has no dependency on jupyter so it's something broken with
your jupyter installation

does it work outside of jupyter in an ipython terminal?

>

moonj94, 2021-10-02:
I tried running in the terminal and there is still an error. It says "Segmentation fault: 11"

agramfort, 2021-10-02:
there is no compiled code in mne so a segfault can only come from

> your installation of python and typically numpy.

I would try to change the version of numpy to see if it helps

moonj94, 2021-10-02:
I downgraded numpy to version 1.20.3 and it did not help.

agramfort, 2021-10-03:
there are some segfault on mac M1 that have been reported on scikit-learn
issue tracker

honestly I don't know

>

moonj94, 2021-10-03:
I am not sure either. When I run the example script here https://mne.tools/stable/auto_examples/decoding/receptive_field_mtrf.html#sphx-glr-auto-examples-decoding-receptive-field-mtrf-py,
the code works fine and I get the result that I expect. But for some reason when it is my own data it doesn't work.

moonj94, 2021-10-03:
@agramfort 
I think the issue might be with the cross validation because if I just do 
`rf.fit(spech,Y)` without using training/testing then the code works just fine. Not sure what the issue is especially since it is working on the example code.

agramfort, 2021-10-05:
can you share a full script that replicates the issue so we can run it on a
different machine?

closed 2021-10-06 by drammock