mne-tools/mne-python #9798: pytest fails to load pooch in developer setup

jbteves, 2021-10-01:
#### Describe the bug
The following Traceback is obtained when attempting to run any form of pytest, such as `pytest mne/tests` from the project root directory, when the current `main` is cloned on my system with the recommended YML-based environment creation and an editable pip install, with all requirements satisfied.
```
ImportError while loading conftest '/Users/tevesjb/repositories/mne-python/mne/conftest.py'.
mne/conftest.py:31: in <module>
    test_path = testing.data_path(download=False)
mne/datasets/testing/_testing.py:22: in data_path
    return _data_path(path=path, force_update=force_update,
mne/datasets/utils.py:213: in _data_path
    pooch = _soft_import('pooch', 'dataset downloading', strict=True)
mne/utils/check.py:273: in _soft_import
    raise RuntimeError(f'For {purpose} to work, the {name} module is '
E   RuntimeError: For dataset downloading to work, the pooch module is needed, but it could not be imported.
```

#### Steps to reproduce
`pooch` is available in the environment that the test is being run from:
```python
>>> import pooch
>>> pooch.__version__
'v1.5.1'
```
Just as bizarrely, if I use the same facility as `check.py`, the load is successful:
```python
>>> import mne
>>> from mne.utils.check import _soft_import
>>> p = _soft_import('pooch', 'dataset downloading', strict=True)
>>> print(p)
<module 'pooch' from '/Users/tevesjb/miniconda3/envs/mne-dev/lib/python3.9/site-packages/pooch/__init__.py'>
```
However, running pytest triggers the above RuntimeError and traceback, regardless of which test (presumably because the error occurs in test configuration).

#### Expected results
The tests at least run, even if they fail.

#### Actual results
The test configuration fails, so I can't even use the debugger via `pytest --trace`.

#### Additional information
```
Platform:      macOS-10.15.7-x86_64-i386-64bit
Python:        3.9.7 | packaged by conda-forge | (default, Sep 23 2021, 07:31:23)  [Clang 11.1.0 ]
Executable:    /Users/tevesjb/miniconda3/envs/mne-dev/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.24.dev0
numpy:         1.21.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.7.0
matplotlib:    3.4.2 {backend=MacOSX}

sklearn:       0.24.2
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.8.0
dipy:          1.4.1
cupy:          Not found
pandas:        1.3.0
mayavi:        4.7.2
pyvista:       0.32.1 {pyvistaqt=0.5.0, OpenGL 4.1 INTEL-14.7.18 via Intel(R) Iris(TM) Plus Graphics OpenGL Engine}
vtk:           9.0.1
PyQt5:         5.12.3
```

larsoner, 2021-10-01:
Can you do `which pytest` and `pytest --version`, as well as `which mne` and verify the output of `python -c "import mne; mne.sys_info()` is the same as  `mne sys_info`? This seems like the sort of thing where you have more than one Python installation on your system, and which `somewhere/bin/pytest` is being run differs from `somewhere_else/bin/python` (but you have MNE installed both places).

You can also add `pooch` to [this list of sys-info'ed modules](https://github.com/mne-tools/mne-python/blob/main/mne/utils/config.py#L543-L545) -- it would actually be great to have this as a PR if you're up for making one!

jbteves, 2021-10-01:
```bash
% which pytest
/Users/tevesjb/miniconda3/envs/mne-dev/bin/pytest
% pytest --version
pytest 6.2.5
% python -c "import mne; mne.sys_info() > python_call.txt
% mne sys_info > cli_call.txt
% diff python_call.txt cli_call.txt
3c3
< Executable:    /Users/tevesjb/miniconda3/envs/mne-dev/bin/python
---
> Executable:    /Users/tevesjb/miniconda3/envs/mne-dev/bin/python3.9
```

I'm up for making a short PR; is there a relevant issue or should I just add it to the referenced tuple?

jbteves, 2021-10-01:
In other news, apparently I was a few days behind and missed that; with the actually up-to-date `main` this bug does not exist; for future reference it appears in 59560dfe1

larsoner, 2021-10-01:
> I'm up for making a short PR; is there a relevant issue or should I just add it to the referenced tuple?

You can just add it to the tuple I think, and update toward the top of the `Enhancements` section of `doc/changes/latest.inc` (with the special "new contributor" substitution + bolding), and also `doc/changes/names.inc` to add your name!

drammock, 2021-10-12:
> apparently I was a few days behind and missed that; with the actually up-to-date `main` this bug does not exist

closing because already fixed

closed 2021-10-12 by drammock