mne-tools/mne-python #9852: Reading Nihon files does not give correct voltage

firejake308, 2021-10-15:
#### Describe the bug
I am using the `mne.read_raw_nihon()` function to read in an EEG file along with its accompanying 21E, LOG, and PNT files. However, when I `plot()` the raw data, some of the channels have voltages that are 10,000x the voltages of the other channels. I tried using a [different tool](https://www.teuniz.net/edfbrowser/) to convert my Nihon file into an EDF, and after opening this EDF, I saw that the voltage amplitudes were much more reasonable. Without knowing too much about Nihon's file format, I'm guessing that there is some sort of `unit_mul`tiplier that MNE isn't reading in that the other tool did notice.

For my personal workflow, I can just use the other tool to generate EDFs and then read them into MNE, but I think it would benefit the project to address this bug.

#### Steps to reproduce
My file contains patient data, so I cannot share it at the moment.

#### Expected results
[image]

#### Actual results
[image]

#### Additional information
```
Platform:      Windows-10-10.0.19041-SP0
Python:        3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:54:40) [MSC v.1900 64 bit (AMD64)]
Executable:    c:\Users\seand\eeg_proj\venv\Scripts\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 9, GenuineIntel: 4 cores
Memory:        7.9 GB

mne:           0.23.4
numpy:         1.19.5 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.5.4
matplotlib:    3.3.4 {backend=Qt5Agg}

sklearn:       0.24.2
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.1.5
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

firejake308, 2021-10-15:
Okay, I think I found the problem! The problematic channels are T7, T8, P7, P8, FT9, and FT10. The reason for this is in the definition of the function `_map_ch_to_specs`, which will use a different digital-to-physical conversion for channel names that are `not in _default_chan_labels`, and apparently these channels that I have are not default. My proposed solution would be to add these channels to the list of default channels, since I think that these labels will always be associated with EEG channels.

610265158, 2024-01-02:
But what is this funtion "_map_ch_to_specs" doing? What is the meaning? I got no clue.

closed 2026-03-18 by larsoner