mne-tools/mne-python #9868: Confusion surrounding Epochs.plot_psd() units

hoechenberger, 2021-10-19:
Consider the following MWE:
```python
# %%
import numpy as np
import matplotlib.pyplot as plt

import mne
from mne.time_frequency import psd_array_multitaper, psd_multitaper
from mne.datasets import somato

data_path = somato.data_path()
subject = '01'
task = 'somato'
raw_fname = op.join(data_path, 'sub-{}'.format(subject), 'meg',
                    'sub-{}_task-{}_meg.fif'.format(subject, task))

raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw, stim_channel='STI 014')

picks = mne.pick_types(raw.info, meg='grad', eeg=False, eog=True, stim=False)
event_id, tmin, tmax = 1, -1., 3.
baseline = (None, 0)
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=baseline, reject=dict(grad=4000e-13, eog=350e-6),
                    preload=True)

epochs.resample(200., npad='auto')  # resample to reduce computation time

# %%
fmin = 1
fmax = 40

epochs.plot_psd(fmin=fmin, fmax=fmax, average=True, dB=True, area_mode=None)
```

This produces the following PSD plot:
[image]

Trying to reproduce this figure manually, I ran:

```python
# %%
psds, freqs = psd_array_multitaper(
    epochs.get_data(),
    sfreq=epochs.info['sfreq'],
    fmin=fmin,
    fmax=fmax,
    n_jobs=8
)

psds_scaled = psds * 1e15**2  # T -> fT
psds_scaled_dB = (10 * np.log10(psds_scaled)).mean(axis=0).mean(axis=0)

_, ax = plt.subplots()
ax.plot(freqs, psds_scaled_dB)
```
This produces:
[image]

The magnitude of the signal clearly differs from what we saw above.

Adjusting the scaling factor to `1e13`, I'm finally able to reproduce the output from `epochs.plot_psd`:

```python
# %%
psds_scaled = psds * 1e13**2  # T -> 100  fT (?!)
psds_scaled_dB = (10 * np.log10(psds_scaled)).mean(axis=0).mean(axis=0)

_, ax = plt.subplots()
ax.plot(freqs, psds_scaled_dB)
```
[image]

Where's my mistake?

cc @cbrnr @agramfort

cbrnr, 2021-10-19:
The original unit is T/m and you are converting to fT/cm, so the factor is 1e13.

guiomar, 2021-10-19:
I think the problem here might be that you are working with gradiometers. 
I was able to reproduce the psd results (with Welch method) for magnetometers. So if you pick `mag` you might be able to get the same plots. Gradiometers may scale differently (maybe due to the `/cm` --> T/m  to fT/cm is 1e13)? You know I opened a few post on the forum related to psd units, so this is the only thing I can tell you ;)

guiomar, 2021-10-21:
Ok. I think I finally understood. I'll explain here in case it can be helpful for others (not sure I'm correct so please go ahead if I say something wrong). I'm new to the tool so I appreciate your patience:

- There's your raw data. With info about the channels with: raw.ch_names (or equivalents)
- There's your epoched data. With info about the channels with: epochs.ch_names (or equivalents)
- However, there's no channel information on the outputs of the psd functions, so one can assume it is the same as the raw or epochs passed as inputs. But this is not always true.

CASE 1: if you have specified the _picks_ parameter inside the psd functions
- In this case an equivalent selection have to be done in the raw/epochs.ch_names list. 
- Since this selection may not be straight forward having just the list of names, something useful could be to create a copy of the epoch just containing the channels you want, e.g. `epochs_grad = epochs.copy().pick('grad')`
- Then you apply the psd function just to the epochs_grad and can ensure that the channels in epochs_grad.ch_names are the ones matching.
- An equivalent thing can be done for the raw. 

CASE 2: _picks_ parameter is not specified or set to None
- By default **the psd is not computed in all the channels passed** e.g. meg_ref channels are excluded, even when the picks parameter is not specified or set to None, as stated in the documentation. So the channels obtained by raw/epochs.ch_names **will not match** the ones obtained with the psd functions. 
- The solution proposed before additionally solves this issue that was making my code crash all the time.

So:
- Pre-selecting your channels before hand creating a copy with .pick (e.g. `epochs_grad = epochs.copy().pick('grad')`) may help to solve this issue, given that the psd functions don't provide channel information on the output.