mne-tools/mne-python #9926: Increase the usage of augmented assignment statements

elfring, 2021-10-29:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
(:point_right: Please check also for questionable change suggestions because of an evolving search pattern.)
```
[Markus_Elfring@fedora lokal]$ perl -p -i.orig -0777 -e 's/^(?<indentation>\s+)(?<target>\S+)\s*=\s*\k<target>[ \t]*(?<operator>[+\-%&|^@]|\*\*?|\/\/?|<<|>>)/$+{indentation}$+{target} $+{operator}=/gm' $(find ~/Projekte/mne-python/lokal -name '*.py')
```

drammock, 2021-10-29:
maybe I'm missing something, but this seems like a pretty low performance / maintainability gain, in exchange for a fair amount of developer time making sure the suggested changes are appropriate.  Personally I think we are in general quite far from needing this degree of optimization, though I'm happy to be convinced otherwise if other devs disagree.

Out of curiosity, did you already run this on our source repo @elfring?  How many lines does it end up changing?

elfring, 2021-10-30:
>…, though I'm happy to be convinced otherwise …

How do you think about to support any collateral evolution according to information from [the Python enhancement proposal 577](https://www.python.org/dev/peps/pep-0577/ "Augmented assignment expressions") (from 2018-05-14)? :thinking:

>…, did you already run this on our source repo … ?

Yes.

:thought_balloon: Under which circumstances would you dare to try an adjusted Perl command variant out on the source files in another test repository?

agramfort, 2021-10-30:
I am not sure to follow what you suggest. I think we use these often to do
inplace operations with numpy and if we don't it's to have a copy (on
purpose).

do I miss something?

elfring, 2021-10-30:
:thought_balloon: Would you like to reconsider any computation details where special programming interfaces like NumPy would be also affected?

agramfort, 2021-10-30:
sorry I don't get what you have in mind

>

cbrnr, 2021-10-30:
As @agramfort and @drammock have already explained, we cannot blindly replace statements like `x = x + 1` with `x += 1`, because their behavior is different. I'm closing this discussion, but if you can explain what exactly you would like to change and why you think this is a good idea we're of course open for any input.

closed 2021-10-30 by cbrnr

elfring, 2021-10-30:
:thought_balloon: Do you care which assignment targets may use in-place operations?

>…, because their behavior is different. …

Can you accept the explanation from the Python documentation?

“…
In the augmented version, `x` is only evaluated once. Also, when possible, the actual operation is performed _in-place_, meaning that rather than creating a new object and assigning that to the target, the old object is modified instead.
…”

>… and why you think this is a good idea we're of course open for any input.

:thought_balloon: Do you accordingly expect positive effects for the software run time characteristics?

agramfort, 2021-10-31:
it's not a matter of speed but of correctness.

when we do

def func(x):
     x = x + 1
     return x

and x is a numpy array we don't modify the input parameter. Blindly
changing:

def func(x):
     x += 1
     return x

would be a disaster as now the function alters its parameters.

>

elfring, 2021-10-31:
>… would be a disaster as now the function alters its parameters.

I suggest to reconsider such a view once more.
:thought_balloon: How many factors do still hinder the wider application of [functionality which became generally available with Python 2](https://docs.python.org/3/whatsnew/2.0.html#augmented-assignment "Augmented assignments")?