mne-tools/mne-python #9932: hide_sensors option does not work in mne.viz.snapshot_brain_montage with pyvista backend

chapochn, 2021-10-30:
### Describe the bug
The option hide_sensors does not have any effect in mne.viz.snapshot_brain_montage with the backend pyvista. The sensors are always shown.

#### Steps to reproduce
in the tutorial:
https://mne.tools/stable/auto_tutorials/clinical/30_ecog.html
up until:
`xy, im = snapshot_brain_montage(fig, raw.info)`

The default value of hide_sensors is True, so there should be no sensors in "im", and there should be no sensor in this image in the tutorial:
https://mne.tools/stable/auto_tutorials/clinical/30_ecog.html#visualize-the-time-evolution-of-the-gamma-power-on-the-brain

#### Expected results
with mayavi backend:
```
xy, im = snapshot_brain_montage(fig, raw.info, hide_sensors=True)
plt.imshow(im)
plt.show()
```
[image]

i.e., no sensors
#### Actual results

with pyvista backend:
```
xy, im = snapshot_brain_montage(fig, raw.info, hide_sensors=True)
plt.imshow(im)
plt.show()
```
[image]

i.e., however, sensors are still there, when they shouldn't

#### Additional information
Platform:      macOS-11.6-x86_64-i386-64bit
Python:        3.9.7 | packaged by conda-forge | (default, Sep 29 2021, 20:33:18)  [Clang 11.1.0 ]
Executable:    /Users/.../opt/anaconda3/envs/mne/bin/python
CPU:           i386: 12 cores
Memory:        16.0 GB

mne:           0.23.4
numpy:         1.21.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.7.1
matplotlib:    3.4.3 {backend=Qt5Agg}

sklearn:       1.0
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.8.1
dipy:          1.3.0
cupy:          Not found
pandas:        1.3.4
mayavi:        4.7.2
pyvista:       0.32.1 {pyvistaqt=0.5.0, OpenGL 4.1 ATI-4.6.20 via AMD Radeon Pro 560X OpenGL Engine}
vtk:           9.0.3
PyQt5:         5.12.3

agramfort, 2021-10-31:
thanks for the bug report. Do you have time to investigate?

>

closed 2021-11-02 by larsoner

chapochn, 2021-11-02:
Thank you for fixing!

PR #9933 (2021-11-02): MRG, BUG: Fix bugs with snapshot_brain_montage
Closes #9931
Closes #9932