mne-tools/mne-python #9939: Error plotting graph

hazmi16194, 2021-11-02:
ValueError                                Traceback (most recent call last)
~\Anaconda3\lib\site-packages\IPython\core\formatters.py in __call__(self, obj)
    339                 pass
    340             else:
--> 341                 return printer(obj)
    342             # Finally look for special method names
    343             method = get_real_method(obj, self.print_method)

~\Anaconda3\lib\site-packages\IPython\core\pylabtools.py in <lambda>(fig)
    246 
    247     if 'png' in formats:
--> 248         png_formatter.for_type(Figure, lambda fig: print_figure(fig, 'png', **kwargs))
    249     if 'retina' in formats or 'png2x' in formats:
    250         png_formatter.for_type(Figure, lambda fig: retina_figure(fig, **kwargs))

~\Anaconda3\lib\site-packages\IPython\core\pylabtools.py in print_figure(fig, fmt, bbox_inches, **kwargs)
    130         FigureCanvasBase(fig)
    131 
--> 132     fig.canvas.print_figure(bytes_io, **kw)
    133     data = bytes_io.getvalue()
    134     if fmt == 'svg':

~\Anaconda3\lib\site-packages\matplotlib\backend_bases.py in print_figure(self, filename, dpi, facecolor, edgecolor, orientation, format, bbox_inches, pad_inches, bbox_extra_artists, backend, **kwargs)
   2191                 print('you pressed', event.button, event.xdata, event.ydata)
   2192 
-> 2193             cid = canvas.mpl_connect('button_press_event', on_press)
   2194         """
   2195 

~\Anaconda3\lib\site-packages\matplotlib\artist.py in draw_wrapper(artist, renderer, *args, **kwargs)
     39         finally:
     40             if artist.get_agg_filter() is not None:
---> 41                 renderer.stop_filter(artist.get_agg_filter())
     42             if artist.get_rasterized():
     43                 renderer.stop_rasterizing()

~\Anaconda3\lib\site-packages\matplotlib\figure.py in draw(self, renderer)
   1861         default = dict(transform=self.transFigure)
   1862 
-> 1863         if (withdash
   1864                 and withdash is not cbook.deprecation._deprecated_parameter):
   1865             text = TextWithDash(x=x, y=y, text=s)

~\Anaconda3\lib\site-packages\matplotlib\image.py in _draw_list_compositing_images(renderer, parent, artists, suppress_composite)
    129     # override the renderer default if suppressComposite is not None
    130     not_composite = (suppress_composite if suppress_composite is not None
--> 131                      else renderer.option_image_nocomposite())
    132 
    133     if not_composite or not has_images:

~\Anaconda3\lib\site-packages\matplotlib\artist.py in draw_wrapper(artist, renderer, *args, **kwargs)
     39         finally:
     40             if artist.get_agg_filter() is not None:
---> 41                 renderer.stop_filter(artist.get_agg_filter())
     42             if artist.get_rasterized():
     43                 renderer.stop_rasterizing()

~\Anaconda3\lib\site-packages\matplotlib\cbook\deprecation.py in wrapper(*inner_args, **inner_kwargs)
    409                 since, message="Passing the %(name)s %(obj_type)s "
    410                 "positionally is deprecated since Matplotlib %(since)s; the "
--> 411                 "parameter will become keyword-only %(removal)s.",
    412                 name=name, obj_type=f"parameter of {func.__name__}()")
    413         return func(*args, **kwargs)

~\Anaconda3\lib\site-packages\matplotlib\axes\_base.py in draw(self, renderer, inframe)
   2705         """
   2706         return self._axisbelow
-> 2707 
   2708     def set_axisbelow(self, b):
   2709         """

~\Anaconda3\lib\site-packages\matplotlib\axes\_base.py in _update_title_position(self, renderer)
   2636 
   2637         # the patch draws the background rectangle -- the frame below
-> 2638         # will draw the edges
   2639         if self.axison and self._frameon:
   2640             self.patch.draw(renderer)

~\Anaconda3\lib\site-packages\matplotlib\axis.py in get_tightbbox(self, renderer, for_layout_only)
   1107                 ilow = view_low
   1108             else:
-> 1109                 # data stops within view, take best tick
   1110                 good_locs = locs[locs <= data_low]
   1111                 if len(good_locs):

~\Anaconda3\lib\site-packages\matplotlib\axis.py in _update_ticks(self)
   1020         # data and view limits.  The unit conversion AxisInfo
   1021         # interface provides a hook for custom types to register
-> 1022         # default limits through the AxisInfo.default_limits
   1023         # attribute, and the derived code below will check for that
   1024         # and use it if is available (else just use 0..1)

~\Anaconda3\lib\site-packages\matplotlib\ticker.py in format_ticks(self, values)
    261 
    262     def format_data(self, value):
--> 263         """
    264         Returns the full string representation of the value with the
    265         position unspecified.

~\Anaconda3\lib\site-packages\matplotlib\ticker.py in <listcomp>(.0)
    261 
    262     def format_data(self, value):
--> 263         """
    264         Returns the full string representation of the value with the
    265         position unspecified.

~\Anaconda3\lib\site-packages\pandas\plotting\_matplotlib\converter.py in __call__(self, x, pos)
    192             return pydt.time(h, m, s, msus).strftime(fmt)
    193         elif ms != 0:
--> 194             return pydt.time(h, m, s, msus).strftime(fmt)[:-3]
    195         elif s != 0:
    196             return pydt.time(h, m, s).strftime("%H:%M:%S")

ValueError: microsecond must be in 0..999999

<Figure size 1440x1440 with 32 Axes>

cbrnr, 2021-11-02:
Please be more specific and tell us what you tried to do and what didn't work. In your traceback, `mne` does not even appear. Ideally, please post a minimal example that reproduces your issue.

hazmi16194, 2021-11-02:
> Please be more specific and tell us what you tried to do and what didn't work. In your traceback, `mne` does not even appear. Ideally, please post a minimal example that reproduces your issue.

I want to plot a multiple graph. It have 5 minutes data interval and i want to produce graph everyday in one section. Here the coding i did.

calendar_dc.tail()

[image]

# define function to multi plot

def multi_plot(data= None, row = None, col = None, title='DC Power'):
    cols = data.columns # take all column
    gp = plt.figure(figsize=(20,20)) 
    
    gp.subplots_adjust(wspace=0.2, hspace=0.8)
    for i in range(1, len(cols)+1):
        ax = gp.add_subplot(row,col, i)
        data[cols[i-1]].plot(ax=ax, style = 'k.')
        ax.set_title('{} {}'.format(title, cols[i-1]))

multi_plot(data=calendar_dc, row=9, col=4)

cbrnr, 2021-11-02:
How is this related to MNE-Python? To me this looks like a pandas and/or matplotlib issue.

agramfort, 2021-11-02:
agreed. it's a pandas / mpl question.

closed 2021-11-02 by agramfort