mne-tools/mne-python #9941: Should channel picking remove unused DigPoint from `info['dig']`

mscheltienne, 2021-11-02:
I don't know if this might have any negative impact somewhere; but I noticed that when you are picking a subset of channels from an instance, and thus removing the others, the DigPoint are not removed from `info['dig']`.

Example with info:

```
ch_names = make_standard_montage('standard_1020').ch_names
info = create_info(ch_names, 512, 'eeg')
info.set_montage('standard_1020')
info.pick_channels(info.ch_names[:10])
```
-> `info['dig']` shows 94 DigPoint for EEG instead of 10.

Example with raw:

```
ch_names = make_standard_montage('standard_1020').ch_names
sfreq = 512
data = np.zeros((len(ch_names), sfreq * 2))
raw = RawArray(data, create_info(ch_names, sfreq, 'eeg'))
raw.set_montage('standard_1020')
raw.pick_channels(raw.ch_names[:10])
```
-> `info['dig']` shows 94 DigPoint for EEG instead of 10.

Is this intended or should `info['dig']` be updated?

agramfort, 2021-11-02:
to me the dig points could still be used for coregistration. At least it
justifies why
you might want to keep it.

>

larsoner, 2021-11-02:
Closing as dup of #8195. Feel free to work on it if you're motivated @mscheltienne !

larsoner, 2021-11-02:
(That issue basically advocates for the opposite, that even when doing `set_montage` we should keep all points rather than remove any, as mentioned by @agramfort here. But the discussion is longer in #8195.)

closed 2021-11-02 by larsoner