mne-tools/mne-python #9961: BUG: 3D functions return non-public members

larsoner, 2021-11-04:
@GuillaumeFavelier we need a public equivalent to mayavi.core.api.Scene for whatever we currently return for create_3d_figure that users can pass around to 3D functions as fig=fig etc. Can you work on this separately? Then we can `git grep "PyVista renderer"` and know all the places we need to put in a reference that links properly.

*adapted from https://github.com/mne-tools/mne-python/pull/9945#issue-772527626*

larsoner, 2022-02-11:
@GuillaumeFavelier this is marked for 1.0, can you take a look?

GuillaumeFavelier, 2022-02-11:
I'll work on it next week :+1:

GuillaumeFavelier, 2022-02-14:
I think the best candidate is the private `_Figure`:

https://github.com/mne-tools/mne-python/blob/main/mne/viz/backends/_pyvista.py#L46

I want to open a small PR to refactor `_PyVistaRenderer` first (mainly moving atrributes into `_Figure`). 

Then we can think about the contract that this `fig` should fulfill. I believe it works as intended already ('saved figure' passed between the 3d functions) but it could be the time to make things right before making it public.

GuillaumeFavelier, 2022-02-15:
> WDYT?

Perfect, that's exactly the purpose of `_Figure`.

larsoner, 2022-02-15:
> Perfect, that's exactly the purpose of _Figure.

Okay cool, I can make a PR to make this public unless you're working on it @GuillaumeFavelier

GuillaumeFavelier, 2022-02-15:
You can go for it @larsoner, it's always possible to refactor things around later on

PR #10347 (2022-02-15): MAINT: Make public Figure3D class
Closes #9961
Closes  #10343

1. I decided to make the abstract base class public, as we can't make `PyVistaFigure3D` for example public, because we can't necessarily import it based on our import mechanics in `backends`. I don't think this is a problem in practice.
2. I also did something a bit weird, which was to make what was in `__init__` go in a new `_init`. This makes it so that we don't have to document all the params in the docstring, and it becomes even clearer that users are not supposed to instantiate directly, and instead should use `create_3d_figur [... cut]
closed 2022-02-15 by larsoner