mne-tools/mne-python #9965: BUG: Tags with spaces are broken

larsoner, 2021-11-05:
This:
```
import mne
import matplotlib.pyplot as plt
report = mne.Report()
figs, captions = list(), list()
for n in range(2):
    fig, ax = plt.subplots()
    ax.text(0, 0, str(n))
    figs.append(fig)
    captions.append(str(n))
report.add_figure(figs, 'numbers', caption=captions, tags=['My numbers'])
report.save('rep.html', overwrite=True)
```
is broken -- the top bar covers content, and the sliders do not work:

[image]

If I change it to `tags['My_numbers']` it works fine:

[image]

@hoechenberger can you look?

hoechenberger, 2021-11-05:
Looks like insufficient tag sanitization, they must not contain any spaces. Will have a look.

larsoner, 2021-11-05:
A nice solution would be to replace spaces with underscores just in the CSS and such and still display visually with spaces, but if that's not possible or too hard, then just raising an error when people have spaces in their tags is okay, too

PR #9970 (2021-11-15): MRG: Add extensive checking for Report tags
Fixes #9965

cc @larsoner
commit b470ddbd (2021-11-15): MRG: Add extensive checking for Report tags (#9970)

* Add extensive checking for Report tags

Fixes #9965

* Update mne/report/report.py

* Apply suggestions from code review [ci skip]
commit 583f366a (2021-11-15): MRG: Add extensive checking for Report tags (#9970)

* Add extensive checking for Report tags

Fixes #9965

* Update mne/report/report.py

* Apply suggestions from code review [ci skip]
closed 2021-11-15 by larsoner