mne-tools/mne-python #9971: mistake in ECoG tutorial

chapochn, 2021-11-06:
Hello, 

In this tutorial:
https://mne.tools/stable/auto_tutorials/clinical/30_ecog.html

it is said: "We can project gamma power from the sensor data to the nearest locations on the pial surface and visualize that:"

However, to my understanding:
```
src = mne.read_source_spaces(
    op.join(subjects_dir, 'fsaverage', 'bem', 'fsaverage-ico-5-src.fif'))
```
reads out sources located on the white matter surface, not the pial surface.

If we want sources on the pial surface, I believe one should first do:
```
src = mne.setup_source_space('fsaverage', subjects_dir=subjects_dir,
                              spacing='ico5', surface='pial')
```

Indeed,
```
src = mne.read_source_spaces(
    op.join(subjects_dir, 'fsaverage', 'bem', 'fsaverage-ico-5-src.fif'))
```
and
```
src = mne.setup_source_space('fsaverage', subjects_dir=subjects_dir,
                              spacing='ico5', surface='white')
```
give the same result.

agramfort, 2021-11-07:
indeed ! either we fix the text or we fix the code

@alexrockhill wdyt?

larsoner, 2021-11-07:
I think we should add a "surface" option that defaults to `'pial'` and just use the vertices (not positions) from the source space unless surface=None

alexrockhill, 2021-11-07:
> I think we should add a "surface" option that defaults to `'pial'` and just use the vertices (not positions) from the source space unless surface=None

Sounds reasonable to me.

larsoner, 2021-11-08:
@chapochn do you want to open a PR to add this behavior? I think we can consider it a bugfix so no need for a deprecation cycle.

closed 2021-11-09 by larsoner

chapochn, 2021-11-09:
Thanks @alexrockhill and @larsoner!

PR #9980 (2021-11-09): [BUG, MRG] Fix projection error
Fixes #9971.

I went ahead and did this one, looks better now I think.