mne-tools/mne-python #9993: mne.stc_near_sensors options desciption

chapochn, 2021-11-10:
Hello,

If I understood different options of mne.stc_near_sensors correctly, I was wondering if in the documentation:

'sum'
    Activation is the sum across each sensor weighted by the fractional distance from each sensor. A sensor with zero distance gets weight 1 and a sensor at distance meters away (or larger) gets weight 0. If distance is less than the distance between any two electrodes, this will be the same as 'nearest'.

should be:

'sum'
    Activation is the sum across each sensor weighted by the fractional distance from each sensor. A sensor with zero distance gets weight 1 and a sensor at distance meters away (or larger) gets weight 0. If distance is less than **half** the distance between any two electrodes, this will be the same as 'nearest'.

larsoner, 2021-11-11:
Have you checked the code @chapochn ? Or maybe tried a case for example where a source is 1mm away from one sensor and 9mm away from another (i.e., less than half the distance) to see if it indeed only gets values from one electrode or if it gets 90% from the first and 10% from the second?

chapochn, 2021-11-12:
Yes, the code and what I tried seem to support that.

What seems sure, is for "nearest", the sources gets the value of sensor **without** any decreases with distance (the line `vals = w[range_, idx] if mode == 'single' else 1.` in the code below), whereas for "sum" and "single", the value decreases with distance (the line `w = np.maximum(1. - dists[vertices] / distance, 0)`). And when we use "single", only the nearest sensor is used (`idx = np.argmax(w, axis=1)`). So "sum" behaves like "single" at a given source only when one or less sensors are within _distance_. But for "sum" and "single" to be the same on a global scale, no point should at a distance smaller than _distance_ from more than 1 sensor. If 2 sources are separated by a distance _distance_, a point between them would be influenced by both sources (because both sources would be at a distance smaller than _distance_). So sources need to be at least 2 _distance_ away. Or _distance_ should be half the minimal distance between sensors/electrodes, so that no "sphere of influence" overlap.

This is the relevant code:

```
    vertices = np.where((dists <= distance).any(-1))[0]
    logger.info(f'    {len(vertices)} / {len(rrs)} non-zero vertices')
    w = np.maximum(1. - dists[vertices] / distance, 0)
    # now we triage based on mode
    if mode in ('single', 'nearest'):
        range_ = np.arange(w.shape[0])
        idx = np.argmax(w, axis=1)
        vals = w[range_, idx] if mode == 'single' else 1.
        w.fill(0)
        w[range_, idx] = vals
    elif mode == 'weighted':
        norms = w.sum(-1, keepdims=True)
        norms[norms == 0] = 1.
        w /= norms
    missing = np.where(~np.any(w, axis=0))[0]
    if len(missing):
        warn(f'Channel{_pl(missing)} missing in STC: '
             f'{", ".join(evoked.ch_names[mi] for mi in missing)}')

    nz_data = w @ evoked.data
```

chapochn, 2021-11-12:
You can try in out with (if you have some _raw_ that has a montage):

```
pos = raw.get_montage().get_positions()['ch_pos']

import scipy.spatial.distance as sdist
xyzs = np.array(list(pos.values()))
distances = sdist.cdist(xyzs, xyzs)

distances = distances + 10000*np.eye(len(distances))
min_dist_between_sensors = np.min(distances)
```

If you set the option _distance_=distance > min_dist_between_sensors/2, then the outputs of mne.stc_near_sensors with options _mode_='single' and _mode_='sum' are not the same. If you set  _distance_=distance <= min_dist_between_sensors/2, then the outputs with options 'single' and 'sum'  are the same.

larsoner, 2021-11-15:
Thanks for investigating, can you make a PR to fix the docstring? Thanks @chapochn !

chapochn, 2021-11-15:
Will do!

PR #10007 (2021-11-16): DOC: stc_near_sensors
#### Reference issue
Fixes #9993

#### What does this implement/fix?
Rectified mistake in the description of the 'sum' option in stc_near_sensors
changed "electrode" to "sensor"
closed 2021-11-16 by larsoner