mne-tools/mne-python #9997: Annotations to_data_frame with all the options similar to Raw.to_data_frame

adam2392, 2021-11-11:
#### Describe the new feature or enhancement
Annotations has a convenience function to go to a data frame, and so does Raw. However, Raw has significantly more flexibility and options.

Now that Annotations also has channel picks, almost all the options that Raw.to_data_frame has can be also added to Annotations.

#### Describe your proposed implementation
Implement Annotations.to_data_frame() with parameters of Raw.to_data_frame().

#### Additional comments
See current Raw.to_data_frame(): https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw

agramfort, 2021-11-11:
why not

>

adam2392, 2021-11-11:
I've been doing a lot of Annotations fancy handling for derived datasets from Raw recently and it's always nice having MNE eat that complexity :p