mne-tools/mne-python #9999: can't read new brainvision markers?

kingjr, 2021-11-12:
When reading a brainvision file (available [here](https://osf.io/5qbr3/?show=revision), after extracting 
- `EEG_data/EEG_excluded.zip/vic0001.vhdr` 
- `EEG_data/hdr_mrk_files/vic0001.vmrk` 
- `EEG_data/hdr_mrk_files/vic0001.vhdr`
)

The following command
```python
raw = mne.io.read_raw_brainvision('vic0001.vhdr')
```

leads to
```
ValueError: MNE-Python currently only supports marker versions 1.0 and 2.0, got unparsable '[Marker Infos]'. 
Contact MNE-Python developers for support.
```

I am thus contacting MNE-Python developers for support :]

cbrnr, 2021-11-12:
Wow, you just won the best issue number jackpot 😄!

It seems like the first line is missing:

```
Brain Vision Data Exchange Marker File, Version 1.0
```

Otherwise, the format looks identical to what we can read, so I'm guessing that this line (1) is either not mandatory or (2) has been removed during export (by BV Analyzer).

cbrnr, 2021-11-12:
ping @sappelhoff WDYT?

cbrnr, 2021-11-12:
According to the official specs, this line is required: file:///Users/clemens/Downloads/BVCDF-1.0_DocVersion-2.5_2019-12-12.pdf (page 12).

kingjr, 2021-11-12:
> Wow, you just won the best issue number jackpot 😄!

haha nice!

larsoner, 2021-11-12:
I guess we could just `warn` in this case that the file might not work correctly. It seems like a bug with whatever wrote this file that the given line is missing

cbrnr, 2021-11-12:
I can do that if no one else beats me to it.

larsoner, 2021-11-12:
Go for it!

PR #10001 (2021-11-12): Warn on missing header in BrainVision files
Fixes #9999.
closed 2021-11-12 by hoechenberger