mne-tools/mne-python #10006: CI: CircleCI now has a 50 minute timeout

larsoner, 2021-11-15:
See https://app.circleci.com/pipelines/github/mne-tools/mne-python/11466/workflows/2cdc35c4-e9cf-4fe6-872a-87a499804609/jobs/38423

I don't know a good solution since it currently takes 2-3h to build our docs.

agramfort, 2021-11-15:
just an idea...

braindecode builds the doc with gh actions but exposes it with circle ci:

https://github.com/braindecode/braindecode/blob/master/.circleci/config.yml

otherwise how much would it cost us?

this one is a real problem....

>

larsoner, 2021-11-15:
> braindecode builds the doc with gh actions but exposes it with circle ci:
>
>https://github.com/braindecode/braindecode/blob/master/.circleci/config.yml

Fine with me to give some variant of this a shot. I can give it a try sometime in the next week or two

> otherwise how much would it cost us?

Based on our usage they recommend $195/mo

[image]

We could probably cut this down quite a bit for example by doing `main` and `maint` builds only weekly (like we do for `linkcheck`) rather than building each merged PR. But this at least gives us some sense of the cost.

sappelhoff, 2021-11-22:
we recently needed to solve that in autoreject as well, see https://github.com/autoreject/autoreject/pull/247, https://github.com/autoreject/autoreject/pull/248, https://github.com/autoreject/autoreject/pull/249

one thing that speeds up builds a lot is changing from the "docker" executor (1 core, 2-4GB RAM):

https://github.com/mne-tools/mne-python/blob/54638887057a2cc771e108e44dd4a61b75de4933/.circleci/config.yml#L18-L19

to a "machine" executor (2 cores, 7.5GB RAM):

https://github.com/autoreject/autoreject/blob/84989d341f0ca93c9d6e0b6f844dfeff0b69b834/.circleci/config.yml#L4-L5

That will probably still not suffice to meet the new 1 hour limit. So another option to stick with CircleCI without having to pay a lot might be to execute "expensive" examples/tutorials in their own jobs (those get their own 60-minute job limit), cache the results, and then stitch it all together.

I had a "hacky" solution for this here in this PR: https://github.com/autoreject/autoreject/pull/247#issuecomment-973069510

We dropped that again, because it turned out that switching from "docker executor" to "machine executor" was enough for the small autoreject docs anyways. --> but it might be an alternative for MNE-Python.

larsoner, 2021-11-22:
Nice, didn't realize it's a limit on each step. Then yes in principle some fancy combination of `make html_dev-pattern` calls could do it!

We talked about this a bit at our dev meeting Friday, as well -- I think we're first going to look to see if we can find a way to pay. We've used CircleCI's services for a long time, so I feel good about (finally!) starting to pay for them if we can make it work.

Then I think should implement nightly-only-if-new-commits building rather than build-after-every-PR. I think this would cut down our usage quite a bit, as the `full` builds are the expensive ones and sometimes we do a bunch in one day. This would only increase the time to deployment from ~4 hours (build + deploy + showing up currently) to at most about a day -- and in practice, probably usually at most less than this, because we schedule the nightly build at a time when we merge few PRs (based on when most merges happen, this would probably be night in Europe and North America).

larsoner, 2021-12-01:
We're paying for CircleCI now, so this can be closed.

closed 2021-12-01 by larsoner