mne-tools/mne-python #10011: T1 and CT do not fuse well after ACPC alignment

leosunpsy, 2021-11-17:
https://mne.tools/stable/auto_tutorials/clinical/10_ieeg_localize.html#sphx-glr-auto-tutorials-clinical-10-ieeg-localize-py
I followed the tutorial above to locate SEEG contacts. I found that the affine registration of the CT with the MR was working optimally only for the original MRI NIfTIfile and CT NIfTI file (created by the MRIcron) or MRI NIfTI file after AC alignment with Freeview (only translation) and CT NIfTI file. But not for MRI NIfTI file after ACPC alignment with Freeview (translation+rotation around the AC point ).

Obviously, the problem is the rotation around AC. Is it that in affine registration the program treats rotation around AC as rotation around the voxel coordinate axis?

I used the related program of Dipy to run again and had the same problem.

[image]
Complete ACPC  Alignment 

[image]
Fusion result After ACPC alignment  with FreeView (only update header)

Python version: 3.9.2
MNE-Python version: 0.24.0
operating system: Debian 11

@alexrockhill

alexrockhill, 2021-11-17:
Really none of those look like amazing alignments. I've been having the same problem with 3/8 patients' data I've been working with very recently. I have manually aligned those three but I'm going to try today to just use the `rigid` without the `translation` as I think this might be causing them to get off to a bad start. Could you try aligning them roughly first and then just running the rigid affine registration (`pipeline=['rigid']`)? I'm very curious to know if that fixes it. I know that requires a manual step but I'm not sure a better solution without modifying the underlying Dipy code.

alexrockhill, 2021-11-18:
You could also try with `zooms=dict(translation=5)` to see if the course translation fixes some kind of local minima where the algorithm is getting stuck.

leosunpsy, 2021-11-18:
> You could also try with `zooms=dict(translation=5)` to see if the course translation fixes some kind of local minima where the algorithm is getting stuck.

Neither approach is feasible.

leosunpsy, 2021-11-18:
I have given up dipy and used ANTsPy (https://github.com/ANTsX/ANTsPy) to align the CT to the MR.

[image]
Fig.4 Fusion result After ACPC alignment

alexrockhill, 2021-11-18:
Looks great, maybe we should consider that instead of what is in MNE currently, thanks for sharing the solution. It might be worth raising an issue on Dipy to see if they can fix their issue.

larsoner, 2021-11-18:
Hmm... it's not on conda-forge yet:

https://github.com/conda-forge/staged-recipes/pull/15000
https://github.com/ANTsX/ANTsPy/issues/213

That being said, I did just `pip install antspx` their 300MB (!) package, so maybe we could make it an optional dependency since even people on `conda-forge` could `pip install antspyx` it.

+1 for making an issue for the dipy folks with a minimal example script + shared data of a case that works with a few `antspy` commands (looks like those are already above) but fails for a few `dipy` commands. If they can fix the problem, that's the best solution. Please mention the issue in this thread once it's open so that others (me included) can subscribe to the dipy issue.

@alexrockhill in the meantime if you want to try adding to `compute_volume_registration` a `backend='dipy' | 'ants'` option that's also a reasonable approach I think. I could imagine different packages doing better/worse on different data, so I think having multiple options for computing the affine isn't a bad idea.

alexrockhill, 2021-11-18:
Just to note, I had a very similar issue with my data and I was able to solve it by pre-aligning and then just using `pipeline=['rigid']` so I would guess maybe the center of mass or translation is causing issues in Dipy.

alexrockhill, 2021-11-18:
I tried the ants registration and it was a lot faster and worked on my data that Dipy failed on. I didn't have to do these lines though

```
warped_img.set_direction(f_img.direction)
warped_img.set_origin(f_img.origin)
warped_img.set_spacing(f_img.spacing)
```

alexrockhill, 2021-11-18:
So ants itself uses a lot of temporary files and has an entirely different image reading system that uses pointers (https://antspy.readthedocs.io/en/latest/core.html?highlight=img_read#ants.core.ants_image.ANTsImage) so I think the only way would be to save any registration images out temporarily and then read them back in with ants. Does this sound like the best way to you @larsoner? I'm just asking because I'm a bit wary that it doesn't quite fit with MNE's ideal style/computation strategy

larsoner, 2021-11-18:
Yes I would just use a temporary directory for that (probably `mne.utils._TempDir`). We have to use it for FreeSurfer commands in `surface.py` and `bem.py` already, so I don't see a (new) problem here really.

alexrockhill, 2022-02-15:
Hey @leosunpsy, in case you weren't aware of developments in the Dipy issue, it seems both ANTSpy and Dipy can get stuck in local minima. One approach might be to try both but what we ended up adding to the tutorial was a manual pre-alignment to avoid getting stuck that takes ~2 minutes. Hope that helps! https://mne.tools/dev/auto_tutorials/clinical/10_ieeg_localize.html

closed 2022-02-15 by alexrockhill