mne-tools/mne-python #10014: Cannot install MNE with conda (have to interrupt "Solving environment: " stage)

neurorepro, 2021-11-17:
#### Describe the bug
When installing a fresh version of MNE, conda hangs on `Solving environment:`

#### Steps to reproduce
```console
conda create -n eeg38 python=3.8
conda activate eeg38
conda install -c conda-forge mne
```

#### Expected results
Conda finishing the installation of MNE and not hanging

#### Actual results
Conda hanging
```console
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment:
```

neurorepro, 2021-11-17:
Update: it could finally succeed after ~ half an hour (although i unfortunately used `--dry-run`...)

Would there be any way to speedup the conda installation ?

drammock, 2021-11-17:
try out [mamba](https://github.com/mamba-org/mamba), and use `mamba install` in place of `conda install`

larsoner, 2021-11-17:
Mixing the conda `main` channel with `conda-forge` is a bit fragile / slow in my experience, perhaps that's one problem? I just did this on Linux:
```
time conda create --channel=conda-forge -n eeg38 python=3.8 mne
```
And solving the environment took ~1 minute on my system (full install took ~2 min), though it did have to go through a "failed" phase as well. Is it any better on your system?

FWIW one that works without any "failed" step is to download and modify our `environment.yml` and do:
```
conda env create -n eeg38 -f ~/python/mne-python/environment.yml
```
But for me this takes about the same amount of time as the command above.

neurorepro, 2021-11-17:
Yes it was much faster to do it in one command. Strange, in the steps i run initially i specifically indicated the `conda-forge` channel (cf initial issue). In any case thanks a lot for the very prompt solution !

closed 2021-11-17 by larsoner