mne-tools/mne-python #10016: MNE - Read Forward Solution from another toolbox

thanosDelatolas, 2021-11-17:
#### Read forward solution
It would be nice to read the leadfield matrix (forward solution) from another toolbox such as [http://duneuro.org/](http://duneuro.org/).

Given the fact that the leadfield matrix is a numpy array, I think that the implementation of this feature is feasible.

#### Additional comments
Maybe this feature wouldn't be necessary if there was a way to solve the forward problem using FEM.

Thank you,

drammock, 2021-11-17:
crossref to forum question: https://mne.discourse.group/t/mne-read-forward-solution-from-another-toolbox/3980

larsoner, 2021-11-17:
In theory we would need more than just the lead field matrix, we'd also need an accompanying `info` and `src` (source space). But this could easily be obtained (hopefully) via something like `info = mne.io.read_raw(...).info`. So in theory we could have an API like:
```
mne.read_forward_solution_duneuro(fname, info, src)
```
And it's up to the user to make sure that `info` and `src` actually reflect the sensor geometry and source space, respectively, used when creating the gain matrix.

It's still unclear to me if we should actually implement this, or if it should be done at the `deneuro` end. @agramfort thoughts?

Either way, @thanosDelatolas as a first step, could you try making a forward solution using `deneuro` from the MNE-Python `sample` dataset, and put the script here? If we really want to support this in MNE-Python, we'll need unit tests that actually read some suitable `duneuro` forward and check that it's similar to the 3-layer BEM solution that we already compute. I think this would be a good first step and even if for now the code for a `read_forward_solution_deneuro` lives in a `gist` it'll allow you to proceed with your analyses.

agramfort, 2021-11-17:
+1 for starting with a gist based on the sample data.

thx !

>

mcpiastra, 2021-11-23:
@thanosDelatolas happy to help in creating the proposed script!

christian-oreilly, 2022-07-14:
@thanosDelatolas @mcpiastra Did you make some progress on this topic or did you drop the matter? I love MNE-Python, but I think the lack of an easy path for FEM-based source reconstruction is currently a bit limitation. Having a way to import FEM models built outside of MNE-Python would be a huge plus.