mne-tools/mne-python #10022: mne.datasets.sample.data_path requires tqdm

mscheltienne, 2021-11-20:
I've been experimenting a bit with GitHub actions on a different repository, and I am using the sample dataset from MNE inside the test. The action is failing because `tqdm` was not installed and is required by `pooch` even if `verbose` is set to None / False. 

```
@verbose
def data_path(path=None, force_update=False, update_path=True, download=True,
              verbose=None):  # noqa: D103

    return _download_mne_dataset(
        name='sample', processor='untar', path=path,
        force_update=force_update, update_path=update_path,
        download=download)
```

Shouldn't there be a mode in which we can download without any verbosity?

drammock, 2021-11-20:
+1 to make it so that download progress bars are suppressed when verbose=False

larsoner, 2021-11-20:
Agreed.

And while we're at it, if you already have the dataset, pooch shouldn't be required. Right now it will throw an error saying you need pooch even if you already have the dataset and don't need to do any downloading.

PR #10210 (2022-01-20): MRG: Fix verbose of mne.datasets fetching
Closes #10022.

Outdated bullet points:

> 
> - [x] Harmonize keyword arguments among all datasets: `force_update`, `update_path`, `download` and `verbose` are now keyword-only everywhere and not only on a subset.
> - [x] Remove `tqdm` dependency for progress bars.
> - [ ] Suppress progress bars if `verbose` is set to `False`.
> - [ ] Remove `pooch` strict dependency if datasets are already downloaded.

----------------------------

**Keyword only arguments**

If the proposal doesn't work for you, I can revert and at least set `verbose` as keyword-only [... cut]
closed 2022-01-20 by larsoner