mne-tools/mne-python #10023: [BUG] Swiss cheese structures in `mne.viz.Brain.add_volume_labels`

alexrockhill, 2021-11-20:
I'm getting structures that have a few holes in them when using the volume labels in `Brain`...
[image]

I wonder if it wouldn't be too hard to implement a binary closing or something of the type but it's in VTK so I'm not really sure what's available there to close structures but I'd assume they would have something like that.

agramfort, 2021-11-20:
it has holes in the 3D grid in the first place?

alexrockhill, 2021-11-20:
It's made from marching cubes of the image so it shouldn't use a grid but it does look like the voxels of the image have hole in their segmentation

agramfort, 2021-11-20:
marching cubes takes a grid as input. you can maybe see if you can add
parameter so
a dilation or something can be done before the marching cubes.

it seems vtk knows how to do things like that:

https://vtk.org/doc/nightly/html/classvtkImageDilateErode3D.html

https://vtk.org/doc/nightly/html/classvtkImageDilateErode3D.html

PR #10024 (2021-11-24): [MRG, ENH] Add `fill_holes` argument to `mne.viz.Brain.add_volume_labels`
Fixes #10023.

I'm a bit stuck, it doesn't look like anything is happening here. Anyone know how to fix this?
closed 2021-11-24 by larsoner