mne-tools/mne-python #10028: Field cancellation maps in a functional label/ROI for EEG/MEG data

dasdiptyajit, 2021-11-23:
#### Describe the new feature or enhancement

Generate field cancellation maps for a functional label/ROI for EEG and MEG data to understand more about  focal vs. widespread cortical patterns. 

additional: The role of depth and orientation of dipoles directly at the sulci and gyri patch to understand their influence in focal vs. widespread source patterns. Some sort of plot that shows dipole orientation for individual label would be a nice feature too. This will give an alternative idea of a source projection to another i.e., mirror sources.   

#### Describe your proposed implementation
A paper to follow up to implement filed cancellation index : https://pubmed.ncbi.nlm.nih.gov/19639553/ 
additional: Some simulations at the source level, followed by some topography maps at the sensor level.   

#### Additional comments
I think this hasn't been implemented in MNE so far. I would be happy to contribute to the proposal but I need someone to guide me as well as for further discussion.

dasdiptyajit, 2021-11-23:
@larsoner something for you to look :)

larsoner, 2021-11-23:
> Some sort of plot that shows dipole orientation for individual label would be a nice feature too.

We have support for plotting source spaces (with orientations I think?), so we could work this in through that. An easy option would be to make it easier to take a source space and restrict it to one Label (e.g., a `mne.source_space.restrict_src_to_label` that works like `forward.restrict_forward_to_label`). Then when you do `src_restricted.plot()` you should get what you want.

A harder option (and maybe less used/useful) would be to allow specifying a parcellation to use to color each source in `src` differently.

> implement field cancellation index ... simulations at the source level, followed by some topography maps at the sensor level.

I'm fine with adding field cancellation index somewhere (`mne.minimum_norm.field_cancellation_index` or `mne.forward.*` depending on if it needs an inv or just a fwd?), and it would be great to have a short example on how to use it. Instead of using simulated data I would just start with `sample` and look for example at if you use some label from `aparc` (e.g., STG) or a much smaller (A1?) label from HCPMMP1 you can see that the cancellation changes or something.

cc @olafhauk you might have ideas for where these should live and/or how they should be implemented and used

dasdiptyajit, 2022-10-18:
Hi @larsoner, 

I am going through this topic again. I have recently started working on this problem. This is what I would like to implement if possible: 

1. Find out: Field cancellation (i.e, net meg/eeg signals) for a given ROI/label, composed with n dipoles (distributed source config, MNE/dSPM) considering there are no activity taking place in the rest of the brain.  //auditory N1 // effect surface curvature to dipole orientation.
Paper:   https://www.sciencedirect.com/science/article/abs/pii/S1053811913001225?fr=RR-1&ref=cra_js_challenge

2. Find out:  Relative amplitude gain change within ROIs ie.e., at least for two ROIs that are activated simultaneously and  placed nearby (e.g., Auditory A1 Vs Insula)         

I need some information regarding some functions which are already existed in MNE simulation model.  Do you think it's a good idea to discuss this here or shall I post this to mne discourse??

larsoner, 2022-10-18:
> I need some information regarding some functions which are already existed in MNE simulation model. Do you think it's a good idea to discuss this here or shall I post this to mne discourse??

The easiest thing if you are working on an actual implementation would be to get as far as you can to the state that something works, or is close to working, but there is something you don't understand, or need but cannot find. Then open a WIP PR and comment in-line on your own PR with questions. Such a "conversation in code" is often easier because it's very concrete and I can see/infer the flow/logic from the code (and you are forced to think about it that way in the process :) )

dasdiptyajit, 2022-10-27:
Yes, I have added this to my to do list. I still need to finish some work with my fMRI data before I can start with this. 
I will tag you once I have some thing working or something I need to know code wise.  cheers!