mne-tools/mne-python #10029: What does "as in nitime" refer to?

guiomar, 2021-11-23:
Hi! I see in many parts of the documentation there is this definition:

> **normalization:** str  
Either “full” or “length” (default). If “full”, the PSD will be normalized by the sampling rate as well as the length of the signal **(as in nitime)**.

What does "as in nitime" refer to? should it be "as in time" or does `nitime` has any special meaning? I've tried to do a search but it only drives me to the different fields were this is used.

drammock, 2021-11-23:
http://nipy.org/nitime/

It's another python package
-------- Original Message --------

guiomar, 2021-11-23:
Oh! Awesome! Thanks :)

guiomar, 2021-11-23:
I have open a very minor PR to specify this. Please ignore it if you think it's obvious.
https://github.com/mne-tools/mne-python/pull/10030

closed 2021-11-23 by guiomar