mne-tools/mne-python #10032: CAR reference as a projector doesn't set custom_ref_applied and remains in projs

mscheltienne, 2021-11-23:
The `.set_eeg_reference` methods have a `projection` argument that can be set to True to set the CAR reference as a projector. I guess in most cases it should be set to False, and the CAR referencing should be directly applied to the data; but if you set it to True, it creates some issues in later processing.

Someone else from my lab encountered two today:

### The flag `custom_ref_applied` is not set to True when the projector is applied.
```
from pathlib import Path

import mne
import numpy as np

directory = Path(mne.datasets.sample.data_path())/'MEG'/'sample'
fname = directory/'sample_audvis_filt-0-40_raw.fif'

#%% CAR
raw = mne.io.read_raw_fif(fname, preload=True)
raw.del_proj()  # 3 PCA, 1 CAR all inactive.
raw.set_eeg_reference('average', projection=False)
data = raw.get_data(picks='eeg')

raw.info['custom_ref_applied']
# Out: 1 (FIFFV_MNE_CUSTOM_REF_ON)

#%% CAR proj
raw = mne.io.read_raw_fif(fname, preload=True)
raw.del_proj()  # 3 PCA, 1 CAR all inactive.
raw.set_eeg_reference('average', projection=True)
data_unapplied = raw.get_data(picks='eeg')
raw.apply_proj()
data_applied = raw.get_data(picks='eeg')

raw.info['custom_ref_applied']
# Out: 0 (FIFFV_MNE_CUSTOM_REF_OFF)

#%% Test
assert np.isclose(data, data_applied).all()
assert not np.isclose(data, data_unapplied).all()
```

Once `.apply_proj` is called, the only way to remove the projector is to delete all channels affected.. i.e. all `eeg` channels in which case neither the projector or the flag `custom_ref_applied` make sense. IMO, it should be set to True when a CAR projector is applied.

### The CAR projector remains in the list of projectors (and becomes active)

This makes probably sense for other types of projectors, but I feel like it's a mistake for the CAR projectors. The changes have been made to the `eeg` data; and the only way to remove it is again to delete all `eeg` channels. 
Now, consider this (sorry I didn't make an example for that one yet, I'll add one tomorrow): 
- I have 2 different EEG recordings from the same participant, same session -> `raw1` and `raw2`. 
- On both, I use `raw.set_eeg_reference('average', projection=True)`. 
- Now, I create similar epochs on both (assume same type of recordings, same events) -> `epochs1` and `epochs2`.
- In `epochs.info['projs']`, the CAR projector (active) is displayed. 
- Now, if I use `mne.concatenate_epochs` to concatenate both in a single epochs instance; an error is raised because projectors between both differ.

This probably makes sense for SSP, PCA projectors; but for a common average reference on EEG channels, it doesn't (at least to me).

For now, to quickly solve the problem my colleague encountered with data at hand (without changing `.set_eeg_reference` proj to False), I proposed to do `epochs.info['projs'] = []`, but obviously I'd like to avoid this.

drammock, 2021-11-23:
projector bookkeeping happens in a lot of places, and changing the behavior of CAR projectors vs other kinds of projectors is (probably?) a very bad idea (or at least a lot of work).  If you really need to concatenate epochs in this way, it's possible to set the CAR *not* as a projector, concatenate your epochs, and then still do inverse imaging:

https://github.com/mne-tools/mne-python/blob/616b9430392d0d2bbf73063e738c7b8c378defcd/mne/io/proj.py#L802-L812

this will detect `custom_ref_applied` and assume your EEG data are zero-mean (just be sure not to delete any channels after applying the CAR... see [this section](https://mne.tools/dev/auto_tutorials/preprocessing/55_setting_eeg_reference.html#eeg-reference-and-source-modeling) of the EEG referencing tutorial for more info).

That said, I can kinda see your point, that once a CAR projector is applied why not set `custom_ref_applied` and remove it from `info['projs']`, and avoid this concatenation issue...  I suspect the reasoning is "don't throw away the additional information in the `Projector` object in case someone wants to interrogate it later, even after it's been applied".  Maybe someone else has a more specific example of why it's kept around?

Of course if you're not ultimately doing inverse imaging, there's (I think?) no reason to want CAR-as-projection anyway, so sticking to `projection=False` makes better sense in that case too, with or without concatenation.

mscheltienne, 2021-11-23:
I forgot about `_needs_eeg_average_ref_proj`, and I thought it was only looking at `'custom_ref_applied'`. That's at least a good thing. 

Of course, I will probably re-preprocess the dataset with CAR _not_ as a projector to get rid of the concatenation issue.
For inverse-mapping, it is recommended to use CAR as a projector in case of channels dropping. But why would you drop channels, especially in a later stage of the processing?

mscheltienne, 2021-11-23:
>  Maybe someone else has a more specific example of why it's kept around?

I'm curious to hear one now 😄 
I have no idea if changing the behavior of CAR projector compared to other kinds of projectors is a good idea, but I don't think it would be a lot of work. The changes should be limited to the function applying the projector?

drammock, 2021-11-23:
> But why would you drop channels, especially in a later stage of the processing?

I wouldn't *recommend* dropping channels late in the preprocessing, it's there as a safeguard I think, in case someone does things out of order.

agramfort, 2021-11-24:
> Now, if I use mne.concatenate_epochs to concatenate both in a single
epochs instance; an error is raised because projectors between both differ.

I don't get this. mne.concatenate_epochs complains even if you did the same
thing for all epochs? if so it's just a bug in mne.concatenate_epochs

>

mscheltienne, 2021-11-24:
From what you are saying, I feel like being able to set CAR as a projection is an unnecessary and complicated safeguard that might even be misleading users..

For `mne.concatenate_epochs`, that was a hard one to reproduce.. The problem is that on this dataset, the following steps are applied:
- Mark bad channels
- Compute CAR (projection = True)
- Interpolate and reset bad channels

Which leads to a mismatch between 2 files on the columns in the CAR projector `['data']['ncol']` and `['data']['col_names']`.

```
from pathlib import Path

import mne

directory = Path(mne.datasets.sample.data_path())/'MEG'/'sample'
fname = directory/'sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(fname, preload=True)
raw.del_proj()
raw.pick_types(eeg=True, stim=True, exclude=[])

#%% Simulate 2 raw instances
raw1 = raw.copy().crop(0, 120)
raw2 = raw.copy().crop(121, None)

#%% CAR
raw1.info['bads'] = []
raw1.set_eeg_reference('average', projection=True)

raw2.info['bads'] = ['EEG 053']
raw2.set_eeg_reference('average', projection=True)
raw2.interpolate_bads(reset_bads=True)

#%% Create epochs
events1 = mne.find_events(raw1, 'STI 001')
events2 = mne.find_events(raw2, 'STI 001')

epochs1 = mne.Epochs(raw1, events1, event_id=5, tmin=0, tmax=0.1,
                     baseline=None, picks='eeg', preload=True)
epochs2 = mne.Epochs(raw2, events2, event_id=5, tmin=0, tmax=0.1,
                     baseline=None, picks='eeg', preload=True)

#%% Concatenate
epochs = mne.concatenate_epochs((epochs1, epochs2))
```

IMO, this specific mismatch should not prevent users from concatenating epochs.

larsoner, 2021-11-24:
> IMO, this specific mismatch should not prevent users from concatenating epochs.

Here I disagree -- the reference for the EEG channels is actually different in the first instance (using all 60 channels) than the second (using all 59 channels).

To me this is actually an argument *for* the projection approach, since it keeps provenance of which channels are actually used in the projection, whereas the non-projection approach would silently allow you to concatenate channels that have the same names but not the same reference.

mscheltienne, 2021-11-24:
>  the reference for the EEG channels is actually different

True, but does it actually matter and impact subsequent analysis? In both cases, it remains an average of the _good_ channels. If I were to first interpolate the bad channels and second compute the CAR projector, I would now have 2 similar projectors computed on the same set of channels. Yet, interpolation will not create any information and in the end, the new reference remains an average of the _good_ channels.

Especially on large-scale studies, it is unlikely that the set of bad channels will match; and thus the average reference used will differ. Does this make the comparison between similar recordings invalid? I don't think so, although you could argue it depends on the distribution of bad channels.

larsoner, 2021-11-24:
> True, but does it actually matter and impact subsequent analysis?

Not sure, but I haven't thought about all possibilities. Thinking about an extreme where one instance has 60 good and another 30 bad and 30 good could get really wacky in terms of the effects (especially if the the bads are spatially concentrated, e.g., all frontal, or all occipital). So I think the answer is "yes, it will matter to varying degrees -- and probably always at least to a tiny degree! -- and depending on the data itself, the channel spatial distribution, and bad channel distribution."

> Yet, interpolation will not create any information and in the end, the new reference remains an average of the good channels.

Note that it will be a different average, though -- it will be a weighted rather than uniform one. The reconstructed channel will be some weighted average of the other 59 channels, so the CAR using all 60 will effectively end up being equivalent to a *non-uniform* weighted average of the original 59 non-interpolated channels, weighted in a way such that it's closer to what you would have gotten if the channel hadn't been bad. So in this sense I wouldn't be surprised if the interpolate-then-concatenate was more valid than the ignore-bad-mismatch-and-concatenate approach.

> Does this make the comparison between similar recordings invalid? I don't think so, although one could argue it depends on the distribution of bad channels.

As with the initial question, I think it's more of a continuum here, not binary valid/invalid.

Maybe a workable solution is add a `on_ref_mismatch='raise'` default that you can set to 'warn' or 'ignore' if you're confident that things won't break for your use case(s) downstream?

mscheltienne, 2021-11-24:
> 60 good and another 30 bad and 30 good could get really wacky

What a terrible recording 😄 But yes, this is only valid if the distribution of bad channels is similar between recordings. 

> Note that it will be a different average, though

Yes, I simplified it a bit. In the case of a low bad channels count, the weighted vs non-weighted CAR would be very close. 

> So in this sense I wouldn't be surprised if the interpolate-then-concatenate was more valid than the ignore-bad-mismatch-and-concatenate approach.

Interpolate then CAR (including interpolated channels) more valid than CAR (excluding bads) then interpolate bads? 
The point is, `raw.interpolate_bads` doesn't affect the CAR projector. Maybe it should?

> Maybe a workable solution is add a on_ref_mismatch='raise' default

It does sound like a good solution, enabling the possibility to concatenate epochs / compare recordings with different ref if you know what you are doing. Personally, for this dataset, I'm just going to set `projection=False` when setting the average reference, thus silently allowing this mismatch.

agramfort, 2021-11-24:
>
> It does sound like a good solution, enabling the possibility to
> concatenate epochs / compare recordings with different ref if you know what
> you are doing.
>

you mean "if you know what you are doing something bad but you don't care.
"  :)

really for me doing this is not clean. For me MNE complains for good
reasons.

mscheltienne, 2021-11-25:
Thanks for the feedback. It now makes sense that MNE complains about this concatenation, or you could get into extreme examples where the references between concatenated epochs are very different. 

However, then what is the use case envisioned for `mne.concatenate_epochs` (for EEG datasets)? I thought it was used to concatenate epochs from multiple similar recordings to help in the analysis. But for EEG, as soon as a CAR is used, it's very unlikely that multiple recordings will have the same CAR reference (even short recordings recorded within a short time frame, e.g. 1h).

To give you an example, my fellow lab member wanted to concatenate the epochs of a task with a significant impact on the alpha-band power from recordings of the same participant/session (5 - 10 short recordings taken within 1h - 1h30); and then pass the concatenation `tfr_morlet` to benefit from the viz capabilities. The concatenation fails because even if the set of bad channels is globally the same, it does vary by 1/2 channels depending on the recording (depending on movements of the participant during this 1h - 1h30). Yet, overall, the CAR reference will not vary much since only a couple of channels are marked as bad and ignored compared to the total number of channels. 

-----------

Also dropping these 2 points again: 
- should `.interpolate_bads` update the CAR projector?
- should `custom_ref_applied` be set to True when a CAR projector is applied? What would be the downside?

larsoner, 2021-11-25:
> However, then what is the use case envisioned for mne.concatenate_epochs (for EEG datasets)?

Usually I use and and see other people use it for concatenating epochs from the same subject, usually from even the same recording.

> should .interpolate_bads update the CAR projector?

It removes entries from `raw.info['bads']` so it would make sense to me that the CAR proj would be updated to include the "new" good channel as well

> should custom_ref_applied be set to True when a CAR projector is applied? What would be the downside?

To me "custom ref applied" means "a non-projector-based reference has been applied" / "do not use or (re)apply an average EEG reference projector".

Historically, MNE originally only ever had support for CAR-as-projector, there was no other reference scheme available -- and there was no `custom_ref_applied` at all. People started asking for *other* referencing methods, so we added them along with this new parameter, basically to have mean the above.

Rather than trying to change what `custom_ref_applied` means, I'd rather put effort toward making it mostly obsolete via something much better and complete like https://github.com/mne-tools/mne-python/issues/8962#issuecomment-788792586. I *think* we came to a consensus on what needed to be done there, we "just" need someone motivated to work on it, as it's non-trivial...

closed 2021-12-10 by mscheltienne