mne-tools/mne-python #10033: Project-wide file overwrite control

wmvanvliet, 2021-11-24:
#### Describe the new feature or enhancement
I would like a project-wide way to control how files are saved. Having to specify `overwrite=True` every time is annoying and makes it lose some of its protection against unwittingly overwriting stuff since it becomes automatic for me to have to write it always. If there is a single master-switch you can toggle to control the saving behavior, that would make life a lot easier.

What would be great is also an option to disable saving altogether. There are times where I'm working on cleaning up a script, or playing around a bit with parameters, or testing someone else's code, without wanting to overwrite the data from my/their "real" run. In this case, I do want the script to continue running (so no error should be raised) but I also don't want any data to be overwritten.

#### Describe your proposed implementation
In every `.save` method, lets change `overwrite=False` to `overwrite='auto'`. The `'auto'` value is controlled through the  `mne.set_save_policy(policy='no-overwrite')` function. The `policy` parameter has three options:

 1. `'overwrite'`: Makes `overwrite='auto'` evaluate to `overwrite=True`
 2. `'no-overwrite'`: Makes `overwrite='auto'` evaluate to `overwrite=False`
 3. `'no-save'`: Disables saving completely. An information/warning message is displayed that saving has been disabled.

#### Describe possible alternatives
Add an option `overwrite='warn'` to the `.save(overwrite=False)` methods. You would still have to specify it every single time, but at least the script would keep running.

agramfort, 2021-11-24:
no objection

are you willing to give it a try?

>

wmvanvliet, 2021-11-24:
> no objection are you willing to give it a try?
Sure!

larsoner, 2021-11-24:
I am a bit worried about someone in a Jupyter notebook setting a save policy then forgetting about it (especially with overwriting files which can pose a big risk to killing data!). One option would be to make this context-manager-only, like:
```
with mne.save_policy('no-save'):
    ...
```
but then the additional indentation could be painful. Instead...

> Add an option overwrite='warn' to the .save(overwrite=False) methods. You would still have to specify it every single time, but at least the script would keep running.

If you refactor your script(s) to have a single new line at the top:
```
overwrite = ...
```
and then everywhere in the given script you use `overwrite=overwrite`, I think you get the behavior you want without any change to MNE (other than maybe adding the `overwrite='warn'` to warn-and-not-save). I'd prefer this solution since I think it's cleaner than adding something new at the MNE end to control a global behavior.

I could imagine MNE having other potential possible global states like `mne.preload_policy(...)` but I'd rather not continue down that road if we can reasonably avoid it. (And for our one existing global state -- `verbose` control -- it at least doesn't usually affect the actual outcome of any script, but rather how much information gets printed along the way.)

agramfort, 2021-11-24:
I would indeed personally do this:

overwrite = ...

foobar(..., overwrite=overwrite)

so I am +0 on this one.

wmvanvliet, 2021-11-26:
That is a good idea. I will start incorporating a `saving` master switch in my analysis projects. We actually don't need any changes on the MNE-Python side if we just do it like this:

```python
if saving:
    raw.save(fname, overwrite=True)
```

closed 2021-11-26 by wmvanvliet